tusen-ai/SST #2: KeyError: 'type

chyohoo, 2022-01-04:
Hi, thanks for this fantastic work.  I have met a problem seeing blow.
```
Traceback (most recent call last):
  File "tools/train.py", line 230, in <module>
    main()
  File "tools/train.py", line 194, in main
    datasets = [build_dataset(cfg.data.train)]
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/datasets/builder.py", line 32, in build_dataset
    build_dataset(cfg['dataset'], default_args), cfg['times'])
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/datasets/builder.py", line 26, in build_dataset
    elif cfg['type'] == 'ConcatDataset':
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/utils/config.py", line 36, in __missing__
    raise KeyError(name)
KeyError: 'type'
```
I have printed the keys in cfg,  but the cfg did have the type key. What could be wrong?

Abyssaledge, 2022-01-04:
Thanks for using SST. I need more information to identify this error: Which config do you use?  Do you modify any codes? And what commands do you run?

chyohoo, 2022-01-04:
Hi, I was trying to use SST on nuScenes radar data. I only modified the config and nuScenes converter. 
and  I ran  this `CONFIG=sst_nuscenes
bash tools/dist_train.sh configs/sst/$CONFIG.py 2 --work-dir ./work_dirs/$CONFIG/ --cfg-options evaluation.pklfile_prefix=./work_dirs/$CONFIG/results evaluation.metric=bbox
` 
Here is the config I use
`Config:
model = dict(
    type='DynamicVoxelNet',
    neck=dict(
        type='SECONDFPN',
        norm_cfg=dict(type='naiveSyncBN2d', eps=0.001, momentum=0.01),
        in_channels=[128],
        upsample_strides=[1],
        out_channels=[384]),
    bbox_head=dict(
        type='Anchor3DHead',
        num_classes=10,
        in_channels=384,
        feat_channels=384,
        use_direction_classifier=True,
        anchor_generator=dict(
            type='AlignedAnchor3DRangeGenerator',
            ranges=[[0.4, -49.6, -1.80032795, 49.6, 49.6, -1.80032795],
                    [0.4, -49.6, -1.74440365, 49.6, 49.6, -1.74440365],
                    [0.4, -49.6, -1.68526504, 49.6, 49.6, -1.68526504],
                    [0.4, -49.6, -1.74440365, 49.6, 49.6, -1.74440365],
                    [0.4, -49.6, -1.74440365, 49.6, 49.6, -1.74440365],
                    [0.4, -49.6, -1.67339111, 49.6, 49.6, -1.67339111],
                    [0.4, -49.6, -1.67339111, 49.6, 49.6, -1.67339111],
                    [0.4, -49.6, -1.61785072, 49.6, 49.6, -1.61785072],
                    [0.4, -49.6, -1.80984986, 49.6, 49.6, -1.80984986],
                    [0.4, -49.6, -1.763965, 49.6, 49.6, -1.763965]],
            sizes=[[1.95017717, 4.60718145, 1.72270761],
                   [2.4560939, 6.73778078, 2.73004906],
                   [2.87427237, 12.01320693, 3.81509561],
                   [2.94046906, 11.1885991, 3.47030982],
                   [2.73050468, 6.38352896, 3.13312415],
                   [0.60058911, 1.68452161, 1.27192197],
                   [0.76279481, 2.09973778, 1.44403034],
                   [0.66344886, 0.7256437, 1.75748069],
                   [0.39694519, 0.40359262, 1.06232151],
                   [2.49008838, 0.48578221, 0.98297065]],
            rotations=[0, 1.57],
            reshape_out=False),
        diff_rad_by_sin=True,
        dir_offset=0.7854,
        dir_limit_offset=0,
        bbox_coder=dict(type='DeltaXYZWLHRBBoxCoder', code_size=7),
        loss_cls=dict(
            type='FocalLoss',
            use_sigmoid=True,
            gamma=2.0,
            alpha=0.25,
            loss_weight=1.0),
        loss_bbox=dict(type='L1Loss', loss_weight=0.5),
        loss_dir=dict(
            type='CrossEntropyLoss', use_sigmoid=False, loss_weight=0.2)),
    train_cfg=dict(
        assigner=[
            dict(
                type='MaxIoUAssigner',
                iou_calculator=dict(type='BboxOverlapsNearest3D'),
                pos_iou_thr=0.5,
                neg_iou_thr=0.3,
                min_pos_iou=0.3,
                ignore_iof_thr=-1),
            dict(
                type='MaxIoUAssigner',
                iou_calculator=dict(type='BboxOverlapsNearest3D'),
                pos_iou_thr=0.5,
[... 473 lines cut ...]
        'max_tokens': 60,
        'drop_range': (30, 60)
    },
    2: {
        'max_tokens': 100,
        'drop_range': (60, 100000)
    }
}, {
    0: {
        'max_tokens': 30,
        'drop_range': (0, 30)
    },
    1: {
        'max_tokens': 60,
        'drop_range': (30, 60)
    },
    2: {
        'max_tokens': 100,
        'drop_range': (60, 100000)
    }
})
shifts_list = [(0, 0), (5, 5)]
fp16 = dict(loss_scale=32.0)

`

chyohoo, 2022-01-04:
the running env is 

> sys.platform: linux
Python: 3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 06:08:21) [GCC 9.4.0]
CUDA available: True
GPU 0,1: Quadro RTX 8000
CUDA_HOME: /usr/local/cuda
NVCC: Cuda compilation tools, release 10.2, V10.2.89
GCC: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
PyTorch: 1.10.1
PyTorch compiling details: PyTorch built with:
  - GCC 7.3
  - C++ Version: 201402
  - Intel(R) Math Kernel Library Version 2020.0.4 Product Build 20200917 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v2.2.3 (Git Hash 7336ca9f055cf1bfa13efb658fe15dc9b41f0740)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - LAPACK is enabled (usually provided by MKL)
  - NNPACK is enabled
  - CPU capability usage: AVX512
  - CUDA Runtime 10.2
  - NVCC architecture flags: -gencode;arch=compute_37,code=sm_37;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_37,code=compute_37
  - CuDNN 7.6.5
  - Magma 2.5.2
  - Build settings: BLAS_INFO=mkl, BUILD_TYPE=Release, CUDA_VERSION=10.2, CUDNN_VERSION=7.6.5, CXX_COMPILER=/opt/rh/devtoolset-7/root/usr/bin/c++, CXX_FLAGS= -Wno-deprecated -fvisibility-inlines-hidden -DUSE_PTHREADPOOL -fopenmp -DNDEBUG -DUSE_KINETO -DUSE_FBGEMM -DUSE_QNNPACK -DUSE_PYTORCH_QNNPACK -DUSE_XNNPACK -DSYMBOLICATE_MOBILE_DEBUG_HANDLE -DEDGE_PROFILER_USE_KINETO -O2 -fPIC -Wno-narrowing -Wall -Wextra -Werror=return-type -Wno-missing-field-initializers -Wno-type-limits -Wno-array-bounds -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-unused-result -Wno-unused-local-typedefs -Wno-strict-overflow -Wno-strict-aliasing -Wno-error=deprecated-declarations -Wno-stringop-overflow -Wno-psabi -Wno-error=pedantic -Wno-error=redundant-decls -Wno-error=old-style-cast -fdiagnostics-color=always -faligned-new -Wno-unused-but-set-variable -Wno-maybe-uninitialized -fno-math-errno -fno-trapping-math -Werror=format -Wno-stringop-overflow, LAPACK_INFO=mkl, PERF_WITH_AVX=1, PERF_WITH_AVX2=1, PERF_WITH_AVX512=1, TORCH_VERSION=1.10.1, USE_CUDA=ON, USE_CUDNN=ON, USE_EXCEPTION_PTR=1, USE_GFLAGS=OFF, USE_GLOG=OFF, USE_MKL=ON, USE_MKLDNN=ON, USE_MPI=OFF, USE_NCCL=ON, USE_NNPACK=ON, USE_OPENMP=ON, 

TorchVision: 0.11.2
OpenCV: 4.5.5
MMCV: 1.4.0
MMCV Compiler: GCC 7.3
MMCV CUDA Compiler: 10.2
MMDetection: 2.14.0
MMSegmentation: 0.14.1
MMDetection3D: 0.15.0+04c8e39

Abyssaledge, 2022-01-04:
Could you please set a breakpoint at
https://github.com/TuSimple/SST/blob/main/tools/train.py#L194
then check the content of `cfg.data.train`?

chyohoo, 2022-01-04:
{'type': 'RepeatDataset', 'data_root': '/home/algo_zf/Data/nuscence/', 'ann_file': '/home/algo_zf/Data/nuscence/nuscenes_infos_train.pkl', 'pipeline': [{'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 18, 'use_dim': 18, 'file_client_args': {'backend': 'disk'}}, {'type': 'LoadPointsFromMultiSweeps', 'sweeps_num': 10, 'use_dim': [0, 1, 2, 3, 5, 8, 9], 'pad_empty_sweeps': True, 'file_client_args': {'backend': 'disk'}}, {'type': 'LoadAnnotations3D', 'with_bbox_3d': True, 'with_label_3d': True}, {'type': 'GlobalRotScaleTrans', 'rot_range': [-0.3925, 0.3925], 'scale_ratio_range': [0.95, 1.05], 'translation_std': [0, 0, 0]}, {'type': 'RandomFlip3D', 'flip_ratio_bev_horizontal': 0.5}, {'type': 'PointsRangeFilter', 'point_cloud_range': [0, -50, -5, 50, 50, 3]}, {'type': 'ObjectRangeFilter', 'point_cloud_range': [0, -50, -5, 50, 50, 3]}, {'type': 'ObjectNameFilter', 'classes': ['car', 'truck', 'trailer', 'bus', 'construction_vehicle', 'bicycle', 'motorcycle', 'pedestrian', 'traffic_cone', 'barrier']}, {'type': 'PointShuffle'}, {'type': 'DefaultFormatBundle3D', 'class_names': ['car', 'truck', 'trailer', 'bus', 'construction_vehicle', 'bicycle', 'motorcycle', 'pedestrian', 'traffic_cone', 'barrier']}, {'type': 'Collect3D', 'keys': ['points', 'gt_bboxes_3d', 'gt_labels_3d']}], 'classes': ['car', 'truck', 'trailer', 'bus', 'construction_vehicle', 'bicycle', 'motorcycle', 'pedestrian', 'traffic_cone', 'barrier'], 'modality': {'use_lidar': True, 'use_camera': False, 'use_radar': False, 'use_map': False, 'use_external': False}, 'test_mode': False, 'box_type_3d': 'LiDAR', 'times': 1, 'dataset': {'load_interval': 10}}

Abyssaledge, 2022-01-04:
In `cfg.data.train`, you have a ` 'dataset': {'load_interval': 10}`, which is wrong. Please refer to https://github.com/TuSimple/SST/blob/main/configs/_base_/datasets/waymo-3d-3class.py#L111
to build the data config.
It would be great if you solve this issue and let me know.

chyohoo, 2022-01-04:
hi, changing the `load_interval  ` value does nothing helpful to the error mentioned above.  But thanks for helping.

Abyssaledge, 2022-01-04:
What I mean is you place the items in wrong place You should fill `cfg.train` like this:
```
    train=dict(
        type='RepeatDataset',
        times=2,
        dataset=dict(
            type=dataset_type,
            data_root=data_root,
            ann_file=data_root + 'waymo_infos_train.pkl',
            split='training',
            pipeline=train_pipeline,
            modality=input_modality,
            classes=class_names,
            test_mode=False,
            # we use box_type_3d='LiDAR' in kitti and nuscenes dataset
            # and box_type_3d='Depth' in sunrgbd and scannet dataset.
            box_type_3d='LiDAR',
            load_interval=1)),
```
`data_root`, `anno_file` ... should be placed under `cfg.train.dataset` instead of `cfg.train`.

Abyssaledge, 2022-01-04:
@chyohoo Does it work now?

chyohoo, 2022-01-04:
> @chyohoo Does it work now?

thanks! It works!!
But another error occurred.

`/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [5,0,0], thread: [32,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.`

Abyssaledge, 2022-01-04:
Could you please provide more detailed error information with CUDA_LAUNCH_BLOCKING=1 ?

chyohoo, 2022-01-04:
2022-01-04 17:48:55,033 - mmdet - INFO - workflow: [('train', 1)], max: 12 epochs
2022-01-04 17:48:55,034 - mmdet - INFO - Checkpoints will be saved to /home/chen_haoye/WS_SST/SST/work_dirs/sst_nuscenes by HardDiskBackend.
drop_info is set to {0: {'max_tokens': 30, 'drop_range': (0, 30)}, 1: {'max_tokens': 60, 'drop_range': (30, 60)}, 2: {'max_tokens': 100, 'drop_range': (60, 100000)}}, in input_layer
/home/chen_haoye/WS_SST/SST/mmdet3d/models/middle_encoders/sst_input_layer.py:321: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  win_coors_x = shifted_coors_x // win_shape_x
/home/chen_haoye/WS_SST/SST/mmdet3d/models/middle_encoders/sst_input_layer.py:322: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  win_coors_y = shifted_coors_y // win_shape_y
No voxel belongs to drop_level:1 in shift 0
No voxel belongs to drop_level:2 in shift 0
/home/chen_haoye/WS_SST/SST/mmdet3d/ops/sst/sst_ops.py:50: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  num_windows = (this_inds // max_tokens).max().item() + 1
/home/chen_haoye/WS_SST/SST/mmdet3d/models/backbones/sst_v1.py:243: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  inv_freq = self.pos_temperature ** (2 * (inv_freq // 2) / pos_length)
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [96,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [97,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [98,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [99,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [100,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [101,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [102,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [103,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [104,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [105,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [106,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [107,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
/opt/conda/conda-bld/pytorch_1639180543123/work/aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [69,0,0], thread: [108,0,0] Assertion `index >= -sizes[i] && index < sizes[i] && "index out of bounds"` failed.
[... cut ...]
frame #23: Py_FinalizeEx + 0x67 (0x55a4cb0dd407 in /home/chen_haoye/anaconda3/envs/SST/bin/python)
frame #24: <unknown function> + 0x248b1b (0x55a4cb0f2b1b in /home/chen_haoye/anaconda3/envs/SST/bin/python)
frame #25: _Py_UnixMain + 0x3c (0x55a4cb0f2e9c in /home/chen_haoye/anaconda3/envs/SST/bin/python)
frame #26: __libc_start_main + 0xe7 (0x7f735355bbf7 in /lib/x86_64-linux-gnu/libc.so.6)
frame #27: <unknown function> + 0x1c761d (0x55a4cb07161d in /home/chen_haoye/anaconda3/envs/SST/bin/python)

Aborted (core dumped)

Abyssaledge, 2022-01-04:
According  to `voxel_size=(0.2, 0.2, 6)` and `point_cloud_range=[0, -50, -5, 50, 50, 3]` in you config, I guess the correct `voxel_size` you want is `(0.2, 0.2, 8)` and the `output_shape` in `SSTv1` should be `(500, 500)`.
Please open another issue if you have a different problem.

chyohoo, 2022-01-04:
Thanks, after setting the correct output_shape, the error disappeared!

closed 2022-01-05 by chyohoo