tusen-ai/SST #4: bboxes1.size(-1) == bboxes2.size(-1)

chyohoo, 2022-01-04:
Hi, i got this error.
Traceback (most recent call last):
  File "/home/chen_haoye/WS_SST/SST/tools/train.py", line 234, in <module>
    main()
  File "/home/chen_haoye/WS_SST/SST/tools/train.py", line 231, in main
    meta=meta)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/apis/train.py", line 34, in train_model
    meta=meta)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/apis/train.py", line 170, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 30, in run_iter
    **kwargs)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/parallel/data_parallel.py", line 75, in train_step
    return self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/models/detectors/base.py", line 237, in train_step
    losses = self(**data)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/detectors/base.py", line 58, in forward
    return self.forward_train(**kwargs)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/detectors/voxelnet.py", line 94, in forward_train
    *loss_inputs, gt_bboxes_ignore=gt_bboxes_ignore)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/dense_heads/anchor3d_head.py", line 354, in loss
    sampling=self.sampling)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/dense_heads/train_mixins.py", line 80, in anchor_target_3d
    sampling=sampling)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/core/utils/misc.py", line 29, in multi_apply
    return tuple(map(list, zip(*map_results)))
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/dense_heads/train_mixins.py", line 148, in anchor_target_3d_single
    gt_labels, input_meta, num_classes, sampling)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/dense_heads/train_mixins.py", line 274, in anchor_target_single_assigner
    gt_bboxes_ignore, gt_labels)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/core/bbox/assigners/max_iou_assigner.py", line 105, in assign
    overlaps = self.iou_calculator(gt_bboxes, bboxes)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/core/bbox/iou_calculators/iou3d_calculator.py", line 46, in __call__
    self.coordinate)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/core/bbox/iou_calculators/iou3d_calculator.py", line 125, in bbox_overlaps_nearest_3d
    assert bboxes1.size(-1) == bboxes2.size(-1) >= 7
AssertionError

Abyssaledge, 2022-01-04:
I believe this error is not related to SST. I can not help you If you do not share your analysis and debugging information.

chyohoo, 2022-01-05:
Hi, the bboxes1 takes value from gt_bboxes_3d in nuscenes dataset which is [N,9], bboxes2 I guess is the detection box which has [N, 7] shape. That caused the AssertionError.

chyohoo, 2022-01-05:
Hi, I found the reason, the nuscene dataset default uses the velocity info in gt_bboxes_3d.

closed 2022-01-05 by chyohoo