tusen-ai/SST #6: mmdetection3d version

MinWang1997, 2022-01-21:
Hi Lue,

Which version of mmdetection 3d did you use? Should I use the same version as you or whichever version is fine?
Thank you!

Best wishes,
Min

Abyssaledge, 2022-01-21:
Our dependencies: 
```
mmcv-full                1.3.8
mmdet                    2.14.0
mmdet3d                  0.15.0
mmpycocotools            12.0.3
mmsegmentation           0.14.1
```
You can use any version of  mmdetection3D as long as having the right dependencies.

ziqipang, 2022-01-22:
@MinWang1997 It seems GitHub will send notifications only when I @ you. Please check if the above answer helps you.

MinWang1997, 2022-01-22:
Dear Ziqi and Lue,

> You can use any version of mmdetection3D as long as having the right dependencies.

Thank you for your reply!

It seems that I should use MMdet3d 0.17.3.

How about torch cuda version etc , since different versions of MMCV compatible to different PyTorch and CUDA versions?

Abyssaledge, 2022-01-23:
@MinWang1997 
I might have not totally understood your needs, while in my opinion, the simplest way to use SST is to clone this repo and install it using `pip install -v -e .` after installing `mmcv`, `mmdet` and `mmsegmentation`. You do NOT need to install the official `mmdet3d`.
I attach our full environment as follows:
```
sys.platform: linux
Python: 3.6.8 (default, Jul  2 2019, 13:27:03) [GCC 5.4.0 20160609]
CUDA available: True
GPU 0,1,2,3,4,5,6,7: NVIDIA GeForce RTX 2080 Ti
CUDA_HOME: /usr/local/cuda
NVCC: Cuda compilation tools, release 10.2, V10.2.89
GCC: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
PyTorch: 1.9.0+cu102
PyTorch compiling details: PyTorch built with:
  - GCC 7.3
  - C++ Version: 201402
  - Intel(R) Math Kernel Library Version 2020.0.0 Product Build 20191122 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v2.1.2 (Git Hash 98be7e8afa711dc9b66c8ff3504129cb82013cdb)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - NNPACK is enabled
  - CPU capability usage: AVX2
  - CUDA Runtime 10.2
  - NVCC architecture flags: -gencode;arch=compute_37,code=sm_37;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_70,code=sm_70
  - CuDNN 7.6.5
  - Magma 2.5.2
  - Build settings: BLAS_INFO=mkl, BUILD_TYPE=Release, CUDA_VERSION=10.2, CUDNN_VERSION=7.6.5, CXX_COMPILER=/opt/rh/devtoolset-7/root/usr/bin/c++, CXX_FLAGS= -Wno-deprecated -fvisibility-inlines-hidden -DUSE_PTHREADPOOL -fopenmp -DNDEBUG -DUSE_KINETO -DUSE_FBGEMM -DUSE_QNNPACK -DUSE_PYTORCH_QNNPACK -DUSE_XNNPACK -DSYMBOLICATE_MOBILE_DEBUG_HANDLE -O2 -fPIC -Wno-narrowing -Wall -Wextra -Werror=return-type -Wno-missing-field-initializers -Wno-type-limits -Wno-array-bounds -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-unused-result -Wno-unused-local-typedefs -Wno-strict-overflow -Wno-strict-aliasing -Wno-error=deprecated-declarations -Wno-stringop-overflow -Wno-psabi -Wno-error=pedantic -Wno-error=redundant-decls -Wno-error=old-style-cast -fdiagnostics-color=always -faligned-new -Wno-unused-but-set-variable -Wno-maybe-uninitialized -fno-math-errno -fno-trapping-math -Werror=format -Wno-stringop-overflow, LAPACK_INFO=mkl, PERF_WITH_AVX=1, PERF_WITH_AVX2=1, PERF_WITH_AVX512=1, TORCH_VERSION=1.9.0, USE_CUDA=ON, USE_CUDNN=ON, USE_EXCEPTION_PTR=1, USE_GFLAGS=OFF, USE_GLOG=OFF, USE_MKL=ON, USE_MKLDNN=ON, USE_MPI=OFF, USE_NCCL=ON, USE_NNPACK=ON, USE_OPENMP=ON, 

TorchVision: 0.10.0+cu102
OpenCV: 4.1.0
MMCV: 1.3.8
MMCV Compiler: GCC 5.4
MMCV CUDA Compiler: 10.2
MMDetection: 2.14.0
MMSegmentation: 0.14.1
MMDetection3D: 0.15.0+5a6bdbc

```

MinWang1997, 2022-01-25:
Sorry for the late reply!
Thank you!

closed 2022-01-25 by MinWang1997

MinWang1997, 2022-02-07:
Hi Ziqi and Lue,
May I ask how many hours do you need for training?
In readme, you said 5-7 hours depends on your devices. Do you mean 5-7 hours by 8*2080Ti? 

Best wishes,
Min

Abyssaledge, 2022-02-07:
Yes, for configs with single frame and 8 heads， it takes 5-7 hours in 8*2080Ti。

发自我的iPhone

MinWang1997, 2022-02-08:
> Yes, for configs with single frame and 8 heads， it takes 5-7 hours in

Hi Lue,

Thank you for your information😊and great work
How about 3 frame?

Best regards,
Min

Abyssaledge, 2022-02-09:
@MinWang1997 . For 3 frames, it takes longer time but it should be less than 10 hours when you use 20% training set and with 8 2080Ti.

Abyssaledge, 2022-02-10:
@MinWang1997 
Do you have any troubles in training speed? To be more clear, for 20% training data, we adopt 12 epoch as schedule.
If you have any different questions, I think opening a new issue is good for community discussions.

MinWang1997, 2022-02-10:
Thanks for your information!
Let me open a new one