tusen-ai/SST #7: Scripts for evaluation

JiaxinZhuang, 2022-01-24:
Hi~Thanks for your amazing work. May I ask can you provide the scrips for evaluation after training from scratch ? Thank you !

Abyssaledge, 2022-01-24:
Thanks for following, here is the evaluation script. You may need to specify config name and work_dirs name.
```
CONFIG=your_config_name
bash ./tools/dist_test.sh configs/work_dirs/$CONFIG.py ./work_dirs/$CONFIG/latest.pth 8 --eval waymo --options "pklfile_prefix=./work_dirs/$CONFIG/results
```

JiaxinZhuang, 2022-01-24:
Thanks your reply! I tried the first command, it rasises the error

`2022-01-24 11:56:01.912434: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 3000000000 Hz
2022-01-24 11:56:01.918235: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x7f58bd4a8290 executing computations on platform Host. Devices:
2022-01-24 11:56:01.918281: I tensorflow/compiler/xla/service/service.cc:175]   StreamExecutor device (0): Host, Default Version
terminate called after throwing an instance of 'std::system_error'
  what():  Resource temporarily unavailable`

When I tried the second command with single GPU, it raises another error. 

`Traceback (most recent call last):
  File "./tools/test.py", line 215, in <module>
    main()
  File "./tools/test.py", line 159, in main
    dataset = build_dataset(cfg.data.test)
  File "/AI/jiaxin/mmdetection3d/mmdet3d/datasets/builder.py", line 41, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/home/jiaxin/anaconda3/envs/open-mmlab/lib/python3.7/site-packages/mmcv/utils/registry.py", line 45, in build_from_cfg
    f'{obj_type} is not in the {registry.name} registry')
KeyError: 'MultiSweepsWaymoDataset is not in the dataset registry'`

Abyssaledge, 2022-01-24:
Which config do you use? Do you modify any codes?

JiaxinZhuang, 2022-01-24:
I didn't modify any codes. Following the getting_started, I have successfully converted and generated the dataset and train the model. Here are the training script. It also fail in the evaluation phase, rasing the same error resource temporarily unavailable
` CONFIG=sst_waymoD5_1x_ped_cyc_8heads_3f
 bash tools/dist_train.sh configs/sst/$CONFIG.py 8 --work-dir ./work_dirs/$CONFIG/ --cfg-options evaluation.pklfile_prefix=./work_dirs/$CONFIG/results evaluation. metric=waymo`

Abyssaledge, 2022-01-24:
For you first error:
It seems related to incorrect setup of `tensorflow` or `waymo-open-dataset`. I could not figure it out with my own environment. I suggest you install the official `mmdet3d` and train a simple `PointPillars` to see if this error occurs.

For the second one:
It is a little bit strange because `MultiSweepsWaymoDataset` should be successfully registered if you can train the model. You can add a breakpoint or `print`  before `class MultiSweepsWaymoDataset` in `waymo_dataset.py` to check if the class is successfully imported when testing.
Let me know if you have any progress or problems.

JiaxinZhuang, 2022-01-24:
May I ask the version of tensorflow and waymo-open-dataset ?

I found the solution for the second error, just enter the SST directory and run command bellow
`
pip install -e .
`
However, I find the since both program will call the function **format_results** in **waymo_dataset.py** , the new error is the same as the first one

Abyssaledge, 2022-01-24:
```
tensorflow               2.4.0
waymo-open-dataset-2-0-0 1.0.1
```
Hope it helps.

JiaxinZhuang, 2022-01-24:
tensorflow-gpu or tensorflow ?  I guess it's the tensorflow-gpu

Abyssaledge, 2022-01-24:
I use tensorflow, because tensorflow-gpu is not really needed in MMDetection3D.  I believe tensorflow-gpu is also fine.

JiaxinZhuang, 2022-01-24:
Finally, although there are still some small error, the output of the evaluation shows below, which looks almost fine ?
`{'Vehicle/L1 mAP': 0.0, 'Vehicle/L1 mAPH': 0.0, 'Vehicle/L2 mAP': 0.0, 'Vehicle/L2 mAPH': 0.0, 'Pedestrian/L1 mAP': 0.798299, 'Pedestrian/L1 mAPH': 0.748522, 'Pedestrian/L2 mAP': 0.718732, 'Pedestrian/L2 mAPH': 0.672069, 'Sign/L1 mAP': 0.0, 'Sign/L1 mAPH': 0.0, 'Sign/L2 mAP': 0.0, 'Sign/L2 mAPH': 0.0, 'Cyclist/L1 mAP': 0.710306, 'Cyclist/L1 mAPH': 0.700095, 'Cyclist/L2 mAP': 0.685552, 'Cyclist/L2 mAPH': 0.675684, 'Overall/L1 mAP': 0.5028683333333334, 'Overall/L1 mAPH': 0.48287233333333335, 'Overall/L2 mAP': 0.4680946666666667, 'Overall/L2 mAPH': 0.449251}`

I use tensorflow instead of tensorflow-gpu as you and sovle the problems. Thanks for you reply!!

Abyssaledge, 2022-01-25:
I think it is a reasonable result. Fun have with SST!

closed 2022-01-25 by JiaxinZhuang