tusen-ai/SST #8: Inf in attached conv layers

chyohoo, 2022-01-26:
Hi, recently I used SST to train on nuScenes dataset. Everything worked fine in the beginning. But after several epchos,  I got nan in bbox_loss and dir_los. And I found that the loss is caused by inf output from the attached conv layer at the end of the sstv1, where the output from recover_bev is fp32 and the intermediate feature maps from conv2d in attached_conv output fp16 values, with the training going on the output values becoming inf. 
I print the weights of conv which is normal.
I tried to clamp the inf value in the feature map, but inf value occurs in the following layers, 
what could be wrong？

happynear, 2022-01-26:
It seems to be a float precision problem. Maybe you can try to run the training in FP32 mode?

chyohoo, 2022-01-27:
> It seems to be a float precision problem. Maybe you can try to run the training in FP32 mode?

thanks for the advice, I am trying to disable fp16 training by commenting [this line](https://github.com/TuSimple/SST/blob/933d8b4a374debaadf84147f027d8ae993353da4/configs/sst/sst_waymoD5_1x_ped_cyc_8heads_3f.py#L119) in the config file.
But I got 
```
Traceback (most recent call last):
  File "./tools/train.py", line 235, in <module>
    main()
  File "./tools/train.py", line 232, in main
    meta=meta)
  File "/home/chen/WS_SST/SST/mmdet3d/apis/train.py", line 34, in train_model
    meta=meta)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/apis/train.py", line 181, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 30, in run_iter
    **kwargs)
  File "/home/che/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/parallel/data_parallel.py", line 75, in train_step
    return self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/models/detectors/base.py", line 237, in train_step
    losses = self(**data)
  File "/home/chene/anaconda3/envs/SST/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/chen/WS_SST/SST/mmdet3d/models/detectors/base.py", line 58, in forward
    return self.forward_train(**kwargs)
  File "/home/chen/WS_SST/SST/mmdet3d/models/detectors/voxelnet.py", line 90, in forward_train
    x = self.extract_feat(points, img_metas)
  File "/home/chen/WS_SST/SST/mmdet3d/models/detectors/dynamic_voxelnet.py", line 50, in extract_feat
    x = self.middle_encoder(voxel_features, feature_coors, batch_size) # SSTInputLayer
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
TypeError: forward() takes 3 positional arguments but 4 were given
```
And then I remove the `auto_fp16` decorator of `SSTInputLayer` and `Base3DDetector` but still get another traceback.:
```
Traceback (most recent call last):
  File "./tools/train.py", line 235, in <module>
    main()
  File "./tools/train.py", line 232, in main
    meta=meta)
  File "/home/chen/WS_SST/SST/mmdet3d/apis/train.py", line 34, in train_model
    meta=meta)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/apis/train.py", line 181, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 30, in run_iter
    **kwargs)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/parallel/data_parallel.py", line 75, in train_step
    return self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/models/detectors/base.py", line 237, in train_step
    losses = self(**data)
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/chen/WS_SST/SST/mmdet3d/models/detectors/base.py", line 58, in forward
    return self.forward_train(**kwargs)
  File "/home/chen/WS_SST/SST/mmdet3d/models/detectors/voxelnet.py", line 90, in forward_train
    x = self.extract_feat(points, img_metas)
  File "/home/chen/WS_SST/SST/mmdet3d/models/detectors/dynamic_voxelnet.py", line 50, in extract_feat
    x = self.middle_encoder(voxel_features, feature_coors, batch_size) # SSTInputLayer
  File "/home/chen/anaconda3/envs/SST/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
TypeError: forward() takes 3 positional arguments but 4 were given
```
Did I miss something？

Abyssaledge, 2022-01-27:
Could you show me your model config?

chyohoo, 2022-01-27:
```
model = dict(
    type='DynamicVoxelNet',
    neck=dict(
        type='SECONDFPN',
        norm_cfg=dict(type='naiveSyncBN2d', eps=0.001, momentum=0.01),
        in_channels=[128],
        upsample_strides=[1],
        out_channels=[384]),
    bbox_head=dict(
        type='Anchor3DHead',
        num_classes=3,
        in_channels=384,
        feat_channels=384,
        use_direction_classifier=True,
        anchor_generator=dict(
            type='AlignedAnchor3DRangeGenerator',
            ranges=[[-49.6, 0, -1.80032795, 49.6, 49.6, -1.80032795],
                    [-49.6, 0, -1.74440365, 49.6, 49.6, -1.74440365],
                    [-49.6, 0, -1.68526504, 49.6, 49.6, -1.68526504]],
            sizes=[[1.95017717, 4.60718145, 1.72270761],
                   [2.4560939, 6.73778078, 2.73004906],
                   [2.87427237, 12.01320693, 3.81509561]],
            rotations=[0, 1.57],
            reshape_out=False),
        diff_rad_by_sin=True,
        dir_offset=0.7854,
        dir_limit_offset=0,
        bbox_coder=dict(type='DeltaXYZWLHRBBoxCoder', code_size=7),
        loss_cls=dict(
            type='FocalLoss',
            use_sigmoid=True,
            gamma=2.0,
            alpha=0.25,
            loss_weight=1.0),
        loss_bbox=dict(type='L1Loss', loss_weight=0.5),
        loss_dir=dict(
            type='CrossEntropyLoss', use_sigmoid=False, loss_weight=0.2)),
    train_cfg=dict(
        assigner=[
            dict(
                type='MaxIoUAssigner',
                iou_calculator=dict(type='BboxOverlapsNearest3D'),
                pos_iou_thr=0.5,
                neg_iou_thr=0.3,
                min_pos_iou=0.3,
                ignore_iof_thr=-1),
            dict(
                type='MaxIoUAssigner',
                iou_calculator=dict(type='BboxOverlapsNearest3D'),
                pos_iou_thr=0.5,
                neg_iou_thr=0.3,
                min_pos_iou=0.3,
                ignore_iof_thr=-1),
            dict(
                type='MaxIoUAssigner',
                iou_calculator=dict(type='BboxOverlapsNearest3D'),
                pos_iou_thr=0.5,
                neg_iou_thr=0.3,
                min_pos_iou=0.3,
                ignore_iof_thr=-1)
        ],
        allowed_border=0,
        code_weight=[1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0],
        pos_weight=-1,
        debug=False),
    test_cfg=dict(
        use_rotate_nms=True,
        nms_across_levels=False,
        nms_pre=4096,
[... 423 lines cut ...]
        'max_tokens': 60,
        'drop_range': (30, 60)
    },
    2: {
        'max_tokens': 100,
        'drop_range': (60, 100000)
    }
}, {
    0: {
        'max_tokens': 30,
        'drop_range': (0, 30)
    },
    1: {
        'max_tokens': 60,
        'drop_range': (30, 60)
    },
    2: {
        'max_tokens': 100,
        'drop_range': (60, 100000)
    }
})
shifts_list = [(0, 0), (5, 5)]
gpu_ids = range(0, 1)

```

chyohoo, 2022-01-27:
I'm guessing the reason why the Inf occurs during the training. I check the log, when the Inf happened, the lr was high (almost 1e-3) at that moment. Probablythe large LR caused the weights becoming larger during bp, and eventually, the feats value go inf in FP16.

Abyssaledge, 2022-01-27:
There is a small difference between:
https://github.com/TuSimple/SST/blob/main/mmdet3d/models/middle_encoders/sst_input_layer.py#L52
and
https://github.com/TuSimple/SST/blob/main/mmdet3d/models/middle_encoders/sst_input_layer_v2.py#L55
Your error should be caused by this difference. 
So I suggest you use config in `sst_refactor`, which is more convenient for you to modify. In `sst_refactor`, we replace `SSTInputLayer` and `SST` with `SSTInputLayerV2` and `SSTv2`.

chyohoo, 2022-01-27:
hi, I updated the files with the v2, I got 
```
Traceback (most recent call last):
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/detectors/dynamic_voxelnet.py", line 35, in __init__
    init_cfg=init_cfg)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/detectors/voxelnet.py", line 34, in __init__
    pretrained=pretrained)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/detectors/single_stage.py", line 32, in __init__
    self.backbone = build_backbone(backbone)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/models/builder.py", line 19, in build_backbone
    return BACKBONES.build(cfg)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/utils/registry.py", line 212, in build
    return self.build_func(*args, **kwargs, registry=self)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/cnn/builder.py", line 27, in build_model_from_cfg
    return build_from_cfg(cfg, registry, default_args)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/utils/registry.py", line 45, in build_from_cfg
    f'{obj_type} is not in the {registry.name} registry')
KeyError: 'SSTv2 is not in the models registry'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tools/train.py", line 235, in <module>
    main()
  File "./tools/train.py", line 193, in main
    test_cfg=cfg.get('test_cfg'))
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/builder.py", line 83, in build_model
    return build_detector(cfg, train_cfg=train_cfg, test_cfg=test_cfg)
  File "/home/chen_haoye/WS_SST/SST/mmdet3d/models/builder.py", line 57, in build_detector
    cfg, default_args=dict(train_cfg=train_cfg, test_cfg=test_cfg))
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/utils/registry.py", line 212, in build
    return self.build_func(*args, **kwargs, registry=self)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/cnn/builder.py", line 27, in build_model_from_cfg
    return build_from_cfg(cfg, registry, default_args)
  File "/home/chen_haoye/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
KeyError: "DynamicVoxelNet: 'SSTv2 is not in the models registry'"
```
do I need to  `pip install -v -e .` again?

chyohoo, 2022-01-27:
tanks, I will try the v2, BTW , have you tried not to use fp16 for training?

Abyssaledge, 2022-01-27:
I do not try fp32 in these configs, but I try fp32 in our customized model and I do not see any problems.

closed 2022-02-07 by chyohoo

Zoenb, 2022-02-22:
@chyohoo  Hi, i also try to use SST to train on nuScenes dataset, could  you please share the results with nuScenes dataset? Thanks!