tusen-ai/SST #10: Questions about the training hyperparameters

1349949, 2022-02-09:
Hi, 
I have some questions about the training hyperparameters.

1. What’s the best model’s training batchsize(samples_per_gpu) exactly is in the Table.2 & 3  of your paper?   All the training configs you provided in the repo set samples_per_gpu=1, is the same as the best model?

2. The ablation study in the paper using 20% data for training.     What about other training hyperparameters?   Such as training epochs, batchsize, training 3 classes together or separately.

I’m doing some reproducing experiments, so I need the training hyperparameters mentioned above.

Abyssaledge, 2022-02-09:
@1349949  Thanks for using!
1. We use `samples_per_gpu=1` for all experiments.
2. For ablation in Table 1, we train 3 classes together for 12 epoch. For other ablations, we train Vehicle and Pedestrian separately. Training 3 classes together will obtain better performances on Pedestrian and Cyclist.

Abyssaledge, 2022-02-11:
@MinWang1997 
1. Although I don't know what you're worried about, you can certainly do it using 20% data.
2. ~ 60h for single frame, maybe 80h for 3 frames with 8 2080Ti.

ziqipang, 2022-02-14:
@MinWang1997 Just to clarify, 20% data for ablation is only valid for Waymo Open Dataset, since 20% of Waymo Open Dataset is still several times larger than KITTI/nuScenes. That said, using 20% data is only applicable and adopted for Waymo Open Dataset and not others.

MinWang1997, 2022-02-15:
Hi Ziqi,

Thank you for your clarification ☺️
Do you think it is also applicable to use 20% Waymo data for normal
training models and tune hyper parameters?

ziqipang, 2022-02-15:
@MinWang1997 Yes, it is valid to tune hyper-parameters with 20% waymo dataset.

MinWang1997, 2022-02-15:
Hi Ziqi,
Thank you for your reply!

closed 2022-02-18 by Abyssaledge