tusen-ai/SST #11: Could you share the 3 class results with 20% training data?

pigtigger, 2022-02-10:
I am curious about the performance gain on vehicle for both 1f && 3f model with 20% training data, 
so could you provide the results , cause I dont have enough GPUs to train it now.

Abyssaledge, 2022-02-10:
The given config `sst_waymoD5_1x_3class_12heads`  could achieve:
```
Vehicle/L1 mAP: 0.6725, Vehicle/L1 mAPH: 0.6670, Vehicle/L2 mAP: 0.5881, Vehicle/L2 mAPH: 0.5832,
Pedestrian/L1 mAP: 0.7346, Pedestrian/L1 mAPH: 0.6057, Pedestrian/L2 mAP: 0.6492, Pedestrian/L2 mAPH: 0.5337,
Sign/L1 mAP: 0.0000, Sign/L1 mAPH: 0.0000, Sign/L2 mAP: 0.0000, Sign/L2 mAPH: 0.0000,
Cyclist/L1 mAP: 0.6321, Cyclist/L1 mAPH: 0.6182, Cyclist/L2 mAP: 0.6086, Cyclist/L2 mAPH: 0.5951,
Overall/L1 mAP: 0.6797, Overall/L1 mAPH: 0.6303, Overall/L2 mAP: 0.6153, Overall/L2 mAPH: 0.5707
```
If you using the 8-head counterpart, Vehicle/L1 mAP should be 1 mAP lower, but I do not have the precise numbers, you could try it if you need the precise numbers.

pigtigger, 2022-02-10:
Thank you! btw, I have some further questions about sst, I wonder if it is convenient to add your social account(wechat is better) for a further discussion. :) @Abyssaledge

Abyssaledge, 2022-02-10:
You could find me by search my github name in wechat.

closed 2022-02-10 by Abyssaledge