tusen-ai/SST #13: What is "save_training" in mmdet3d/datasets/waymo_dataset.py format_results

Hiusam, 2022-02-24:
Hi, I have checked the official code of OpenMMLab and there is no "save_training" attribute. What does it mean? This attribute will cost prefix=0 and cannot find my validation bin files, which start with "1".

``
if self.split == 'training':
    if self.save_training:
        waymo_tfrecords_dir = osp.join(waymo_root, 'training')
                prefix = '0
    else:
        waymo_tfrecords_dir = osp.join(waymo_root, 'validation')
        prefix = '1'
``

Abyssaledge, 2022-02-24:
We add this attribute to create `xxx.bin` files on training set, which is needed by `LiDARRCNN`. I guess there will be no problems if you set it to `False`.

Hiusam, 2022-02-24:
In your data/waymo/kitti_format/validation/velodyne, the bin files begin with 0 or 1? When I test the model, the terminal will log some info like "1039064 not found". These infos are triggered in prediction_kitti_to_waymo.py/convert_one. Do you know why and does this affect the result?

Abyssaledge, 2022-02-24:
1. prefix `0` for training set, `1` for validation set and `2` for testing set, which is determined by official mmdet3d data preparation.
2. It is normal behavior that some logs like`1039064 not found` appear in the terminal occasionally, which means the detector have no prediction in this frame.  You are very likely to see that when your detector has very high precision. It has no negative effect.

closed 2022-02-24 by Hiusam