tusen-ai/SST #14: Questions about SSTInputlayer

JiaxinZhuang, 2022-02-25:
Hi~ I run run.sh, its configs use SSTInputlayer, which is fed with three parameters as below. However, the definition of forward function for SSTInputlayer (not SSTInputlayer v2) only contains two parameters. The code ran well without any error, which I can't figure out why.

[image]

[image]

Abyssaledge, 2022-02-25:
Hi, thanks for using. I use `SSTInputLayerV2` In `run.sh` in current version. I wonder which config do you use in your `run.sh`?

JiaxinZhuang, 2022-02-25:
Thanks for your reply, I used sst_waymoD5_1x_ped_cyc_8heads_3f.

closed 2022-03-01 by JiaxinZhuang