tusen-ai/SST #15: Training config for paper results

Stephanie-Shen324, 2022-03-07:
Would you mind sharing the configs used for table 2 & 3? Since they are trained for 24 epochs on the full waymo dataset, which takes a long time to run, I want to get it right for reproducing the results.

Thank you very much for the help!

Abyssaledge, 2022-03-07:
Thanks for using SST.
I will update the configs later.
If you are in a hurry, you could refer to `sst_waymoD5_1x_car_8heads_wnms.py` and `sst_waymoD5_1x_ped_cyc_8heads_3f.py`. You just need to change the epoch number to 24 and the dataset interval to 1. If it goes well you will get similar numbers in the tables or even better.

Stephanie-Shen324, 2022-03-07:
Thank you for the prompt reply!

How about the learning rate setup? I assume you used cosine_2x.py, where the lr is 1e-5, while in the paper the lr is 0.001.

Abyssaledge, 2022-03-07:
You don't need to change the learning rate, 1e-5 is the initial learning rate before warmup, which will increase to 1e-3 after 10% iterations.

Stephanie-Shen324, 2022-03-07:
Got it. And did you use the Weighted NMS for car?

Abyssaledge, 2022-03-07:
> Got it. And did you use the Weighted NMS for car?

Yes, it brings around 0.6 AP improvement in SST_1f.

Stephanie-Shen324, 2022-03-07:
Thank you! I will update results when I get them. Great repo by the way!

closed 2022-03-07 by Stephanie-Shen324