tusen-ai/SST #17: hyperparameter window_shape

Stephanie-Shen324, 2022-03-08:
Hi,
I have a few questions and hope that you can answer them:

1. I noticed that the window_shape hyperparameter is (10,10) for ped_cyc training and (12,12) for car and 3 classes training, is this observation true?

2. (10,10) means each window has 10*10*1 pillars right?

3. what is the meaning of the hyperparameter drop_info? what does 0,1,2 in the dictionary keys represent?

Abyssaledge, 2022-03-08:
1. Yes. It's right.
2. (10, 10) means each window has **at most** 100 pillars (voxels), but the number of voxels in each window is dynamic.
3. We use bucketing-and-padding technique. The drop_info contains related information. For clear understanding, you could refer to our paper, where we name it as `batching`. And  the following Chinese blog may help users to understand this technique: https://www.cnblogs.com/callyblog/p/9681377.html

Stephanie-Shen324, 2022-03-08:
Thank you for the replies! For drop_info_testing, you set three buckets for ped_cyc and four buckets for cars and 3 classes?

Abyssaledge, 2022-03-08:
I believe it is right. You do not need to worry about this.