tusen-ai/SST #19: How to viz the learned sparse regional attention?

maudzung, 2022-03-15:
Dear authors,

Thanks for your interesting paper.
Can you please share with us the source code that uses for the visualization of the learned sparse regional attention? I'd like to reproduce `Fig. 5` in your paper.
Thank you so much!

Abyssaledge, 2022-03-15:
Thanks for using SST.
I'm sorry that we can not provide you with our visualization code right now because the viz code is quite dirty and incompatible with the current codebase.
Basically, we first save the attention weights of the last block and some related position information (e.g., voxel coordinates, window ID). Then we randomly pick queries in foreground objects and paint the point cloud with the corresponding attention weights in Open3D.

maudzung, 2022-03-18:
Thank you so much! 👍

closed 2022-03-18 by maudzung

yinheyanxian, 2024-06-24:
> Thank you so much! 👍

Hello, have you realized the visualization of the learned sparse regional attention? I also want to draw a picture like Figure 5.