tusen-ai/SST #22: Hyperparameter settings in SST

lihao0374, 2022-03-31:
Great job.

If I want to try SST on a dataset with low density （LiDAR with 32 lasers）, how should I modify the  hyperparameters in SST.
If I use a smaller voxel size (0.16, 0.16, 6), how should I adjust the hyperparameters in SST, such as drop info, window shape settings and so on? 

I am looking forward to your reply.

Abyssaledge, 2022-03-31:
Although I have not tried this setting, my suggestions are (in the refactored version):
```
window_shape = (20, 20, 1) ~ (24, 24, 1)
drop_info  = {
    0:{'max_tokens':30, 'drop_range':(0, 30)},
    1:{'max_tokens':60, 'drop_range':(30, 60)},
    2:{'max_tokens':100, 'drop_range':(60, 100)},
    3:{'max_tokens':150, 'drop_range':(100, 1000)},
}
```
Remember to change the `sparse_shape` according to your point cloud range and voxelization size.
And if I were you, I might draw the distribution of the number of pillars in a window to determine the drop_info.

lihao0374, 2022-03-31:
> Thank you for your reply. 

In addition, I notice that only 'x,y,z' of points are used during training, and the intensity information in Waymo dataset is not used. What is the reason for this？

Abyssaledge, 2022-03-31:
The intensity in Waymo is not normalized, using them directly may lead to NaN output. It's fine to use it If you clip or normalize it into a reasonable range.

closed 2022-04-01 by lihao0374