tusen-ai/SST #28: Exception has occurred: NvvmError

cjfghddl0206, 2022-05-19:
When I run python tools/train.py and epoches 1 runs and then moves to epoches 2, an error stating nvvmerror: failed to compile occurs. I installed numba to solve this, but the error persists.

Traceback (most recent call last):
  File "tools/train.py", line 238, in <module>
    main()
  File "tools/train.py", line 235, in main
    meta=meta)
  File "/home/pch/SST_kitti/mmdet3d/apis/train.py", line 34, in train_model
    meta=meta)
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/apis/train.py", line 170, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 54, in train
    self.call_hook('after_train_epoch')
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/base_runner.py", line 307, in call_hook
    getattr(hook, fn_name)(self)
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/hooks/evaluation.py", line 220, in after_train_epoch
    self._do_evaluate(runner)
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/mmdet/core/evaluation/eval_hooks.py", line 19, in _do_evaluate
    key_score = self.evaluate(runner, results)
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/mmcv/runner/hooks/evaluation.py", line 311, in evaluate
    results, logger=runner.logger, **self.eval_kwargs)
  File "/home/pch/SST_kitti/mmdet3d/datasets/kitti_dataset.py", line 355, in evaluate
    self.CLASSES)
  File "/home/pch/SST_kitti/mmdet3d/core/evaluation/kitti_utils/eval.py", line 707, in kitti_eval
    eval_types)
  File "/home/pch/SST_kitti/mmdet3d/core/evaluation/kitti_utils/eval.py", line 613, in do_eval
    min_overlaps)
  File "/home/pch/SST_kitti/mmdet3d/core/evaluation/kitti_utils/eval.py", line 479, in eval_class
    rets = calculate_iou_partly(dt_annos, gt_annos, metric, num_parts)
  File "/home/pch/SST_kitti/mmdet3d/core/evaluation/kitti_utils/eval.py", line 382, in calculate_iou_partly
    dt_boxes).astype(np.float64)
  File "/home/pch/SST_kitti/mmdet3d/core/evaluation/kitti_utils/eval.py", line 116, in bev_box_overlap
    from .rotate_iou import rotate_iou_gpu_eval
  File "/home/pch/SST_kitti/mmdet3d/core/evaluation/kitti_utils/rotate_iou.py", line 292, in <module>
    criterion=-1):
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/numba/cuda/decorators.py", line 101, in kernel_jit
    kernel.bind()
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/numba/cuda/compiler.py", line 548, in bind
    self._func.get()
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/numba/cuda/compiler.py", line 426, in get
    ptx = self.ptx.get()
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/numba/cuda/compiler.py", line 397, in get
    **self._extra_options)
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/numba/cuda/cudadrv/nvvm.py", line 496, in llvm_to_ptx
    ptx = cu.compile(**opts)
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/numba/cuda/cudadrv/nvvm.py", line 233, in compile
    self._try_error(err, 'Failed to compile\n')
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/numba/cuda/cudadrv/nvvm.py", line 251, in _try_error
    self.driver.check_error(err, "%s\n%s" % (msg, self.get_log()))
  File "/home/pch/anaconda3/envs/SST/lib/python3.7/site-packages/numba/cuda/cudadrv/nvvm.py", line 141, in check_error
    raise exc
numba.cuda.cudadrv.error.NvvmError: Failed to compile

<unnamed> (66, 23): parse expected comma after load's type
NVVM_ERROR_COMPILATION

Abyssaledge, 2022-06-06:
Sorry for the late reply. It seems that this problem occurs when you evaluate the model after the first epoch. I guess it is related to improper installation of numba. You could use a simple unittest to verify the installation of numba.

cjfghddl0206, 2022-06-07:
I already installed numba. 
This version was 0.48.0.

closed 2022-06-25 by Abyssaledge