tusen-ai/SST #30: question about position embedding on query, key, value

seonhoon1002, 2022-06-03:
Hello, I have a question about position embedding on query, key, value

In your paper, F'=MSA(LN(F),PE(L))+F, so I guessed that the position embedding is added to q,k,v samely. 

But in your codes, postion embedding is only added to query and key, not value.

So I wonder why you add position embedding seperately.

Abyssaledge, 2022-06-06:
In my opinion, using PE in queries and keys is theoretically correct, because the similarity is calculated based on the query and key, then the queries and keys have larger similarities when they are spatially close (with similar PE).
However, I believe it is also right to add PE to values, you could try it.

seonhoon1002, 2022-06-06:
Thank you for your reply

closed 2022-06-06 by seonhoon1002