tusen-ai/SST #31: Is 'ingroup_indices' a CUDAExtension?

Devoe-97, 2022-06-06:
Very excited to see the updated code that supports CenterHead!
I was wondering if ingroup_indices is a CUDAExtension?
https://github.com/TuSimple/SST/blob/b3588728f63c59413ef6e9cf2f0bda1f58396220/mmdet3d/ops/sst/sst_ops.py#L197

Abyssaledge, 2022-06-06:
Yes, see the last paragraph of Usage in README. Let me know if you have further questions.

Devoe-97, 2022-06-06:
> Yes, see the last paragraph of Usage in README. Let me know if you have further questions.

I see it. Thanks!

closed 2022-09-21 by Abyssaledge

WYRTDCQ, 2023-09-05:
> Hi, @Abyssaledge @Devoe-97 @happynear I met the same error. I don't find the readme about how to solve it.
> 
> ```
>   File "/defaultShare/SHFP12/xiaoquan.wang/01_bev/SST/mmdet3d/ops/sst/sst_ops.py", line 244, in <module>
>     import ingroup_indices
> ```
I don't find too.Do you solve it?

Abyssaledge, 2023-09-06:
@WYRTDCQ  Please follow the instructions first: https://github.com/tusen-ai/SST/blob/main/docs/overall_instructions.md#basic-usage
Specifically, you should install TorchEx.