tusen-ai/SST #32: How to set the parameters of drop_info to make the model more efficient when applying SST on other datasets?

Devoe-97, 2022-06-08:
Hi, I would like to know how to set the parameters of drop_info to make the model more efficient when applying SST on other datasets? 
It appears that the parameters of drop_info have an impact on model efficiency and performance?

Abyssaledge, 2022-06-08:
It has no impact on the performance (e.g. mAP) as long as you do not change the max number of tokens in the last level of drop_info.
But it has an impact on the memory footprint. From my experience,  in some datasets, it might be better to lower to max_tokens in each level of drop_info for memory efficiency.
For example, if your data is more sparse than waymo, you may change the default one to 
```
drop_info_training ={
    0:{'max_tokens':15, 'drop_range':(0, 15)},
    1:{'max_tokens':30, 'drop_range':(15, 30)},
    2:{'max_tokens':60, 'drop_range':(30, 60)},
    3:{'max_tokens':100, 'drop_range':(60, 10000)},
}
```
or
```
drop_info_training ={
    0:{'max_tokens':20, 'drop_range':(0, 20)},
    1:{'max_tokens':40, 'drop_range':(20, 40)},
    2:{'max_tokens':80, 'drop_range':(40, 10000)},
}
```
to see if it is better.

And if you frequently receive the messages: 'No voxel belongs to drop_level xxx', it suggests you can further lower the max_tokens in drop_info.

Devoe-97, 2022-06-08:
In my dataset, reducing max_tokens will drop a lot of voxels, because many windows have more than 200 voxels. should I reduce the window size to reduce memory usage in this case? Will reducing the window size affect the performance?

Abyssaledge, 2022-06-08:
I suggest adopting the default pillar size and window size in our configs at first. This setting is verified effective in several internal datasets. In my opinion, it is a good idea to tune these hyper-parameters after achieving acceptable performance with the default setting.

Devoe-97, 2022-06-08:
> I suggest adopting the default pillar size and window size in our configs at first. This setting is verified effective in several internal datasets. In my opinion, it is a good idea to tune these hyper-parameters after achieving acceptable performance with the default setting.

Thanks, I will give it a try.

closed 2022-06-08 by Devoe-97