tusen-ai/SST #33: pretrained weights about backbone?

gp1234567, 2022-06-08:
1. Did you release the pre training weight of backbone？
2. I used two frames of data to superimpose and input on the 128 line lidar. The results of 1.3W data after 20epoch are as follows. Can you give me some comments?
[image]

Abyssaledge, 2022-06-08:
1. We can not release the pretrained model due to the license of Waymo.
2. Could you present more details about your experiment? Which config do you use? Any modifications?

gp1234567, 2022-06-08:
My config is as follows
class_names = [['Car', 'Truck'], ['Pedestrian', 'Cyclist']]
class_names_list = ['Car', 'Truck', 'Pedestrian', 'Cyclist']
voxel_size = [0.16, 0.16, 5]
det_range = [-76.8, -39.68, -1.0, 102.4, 39.68, 4]
grid_size = [1120, 496, 1]
sub_grid_size = [1120, 496, 5]
max_points_per_voxel = 32
num_point_features = len(use_dim)

window_shape=(24, 24, 1) # 24 * voxel_size[0]
drop_info_training ={
    0:{'max_tokens':30, 'drop_range':(0, 30)},
    1:{'max_tokens':60, 'drop_range':(30, 60)},
    2:{'max_tokens':100, 'drop_range':(60, 100)},
    3:{'max_tokens':150, 'drop_range':(100, 100000)},
}
drop_info_test ={
    0:{'max_tokens':30, 'drop_range':(0, 30)},
    1:{'max_tokens':60, 'drop_range':(30, 60)},
    2:{'max_tokens':100, 'drop_range':(60, 100)},
    3:{'max_tokens':150, 'drop_range':(100, 100000)},
}
drop_info = (drop_info_training, drop_info_test)
shifts_list=[(0, 0), (window_shape[0]//2, window_shape[1]//2) ]

train_cfg = None
test_cfg = dict(
    nms_type='nms_gpu',
    nms_thresh=0.1,
    score_thresh=0.3,
    use_raw_score=True,
    nms_pre_maxsize_last=4096,
    nms_post_maxsize_last=500,
    out_size_factor=1,
    voxel_size=voxel_size,

    box_filter=dict(
        use_image_area_filter=True,
        limit_range=det_range,
    )
)
ious_config =dict(
    pc_range=det_range,
    voxel_size=voxel_size,
    out_size_factor=1,
    type='iou_3d',
)
label_assignment = dict(
    out_size_factor=1,
    dense_reg=1,
    gaussian_overlap=0.1,
    max_objs=500,
    min_radius=2,
    num_classes=class_names_list.__len__(),
    vel=True,
    motion_state=True,
    grid_size=grid_size,
    voxel_size=voxel_size,
    pc_range=det_range,
    ignore_direction=ignore_direction,
    ignore_pedestrian_angle=ignore_pedestrian_angle,
    ignore_far_pedestrian=ignore_far_pedestrian,
)

model = dict(
    type='SST_CP_MULTIPILLAR',
    vfe=dict(
        type='PillarFeatureNet',
        voxel_size=voxel_size,
        pc_range=det_range,
        use_norm=True,
        num_filters=[128],
        with_distance=False,
    ),
    mid_encoder=dict(
        type='SSTInputLayerV2',
        window_shape=window_shape,
        sparse_shape=(496, 1120, 1),
        shuffle_voxels=True,
        debug=True,
        drop_info=drop_info,
        pos_temperature=10000,
        normalize_pos=False,
    ),
    backbone=dict(
        type='SSTv2',
        d_model=[128, ] * 6,
        nhead=[8, ] * 6,
        num_blocks=6,
        dim_feedforward=[256, ] * 6,
        output_shape=[496, 1120],
        num_attached_conv=3,
        conv_kwargs=[
            dict(kernel_size=3, dilation=1, padding=1, stride=1),
            dict(kernel_size=3, dilation=1, padding=1, stride=1),
            dict(kernel_size=3, dilation=2, padding=2, stride=1),
        ],
        conv_in_channel=128,
        conv_out_channel=128,
        debug=False,
    ),
    neck = dict(
        type='SECONDFPN',
        in_channels=[128, ],
        upsample_strides=[1, ],
        out_channels=[384, ]
    ),
    head=dict(
        type='SSTHead',
        cfg=dict(
            task_names=class_names,
            grid_size=np.array(grid_size),
            box_loss_list=box_loss_list,
            in_channels=384,  # 128 + 128 + 128
            shared_conv_channel=64,

            head_cfg=dict(
                head_in_channels=64,
                head_conv=64,
                final_kernel=3,
                bn=True,

                heads=[
                    dict(
                        hm=[2, 3],
                        reg=[2, 3],
                        height=[1, 3],
                        dim=[3, 3],
                        rot=[2, 3],
                        vel=[2, 3],
                        ious=[1, 3],
                        state=[1, 3]
                    ),
                    dict(
                        hm=[2, 3],
                        reg=[2, 3],
                        height=[1, 3],
                        dim=[3, 3],
                        rot=[2, 3],
                        vel=[2, 3],
                        ious=[1, 3],
                        state=[1, 3]
                    )
                ],
            )
        ),
        loss=dict(
            code_weights=[
                [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.25, 0.25, 0.5, 0.5, 0.5],
                # [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.5, 0.5],
                [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.5, 0.5, 0.5],
            ],
            vel=True,
            motion_state=True,
            hm_weight=[2.0, 2.0],
            # hm_weight=[1.0, 1.0],
            loc_weight=[1.0, 0.5],
            # loc_weight=[2.0, 2.0],
            box_weight=[0.25, 0.0],
            # box_weight=[0.0, 0.0]
            iou_loss=True,
            ious_config=ious_config
        )
    )
)

gp1234567, 2022-06-08:
I use sst_v2 backbone + detect_head of anchor free，like centerpoint

Abyssaledge, 2022-06-08:
I think the pillar size (0.16m) is too small for CenterHead, because the object center is hard to be completely filled with several convolutions after the SST backbone.
I suggest use pillar of 0.32m and follow the recently updated `configs/sst_refactor/sst_waymoD5_1x_3class_centerhead.py`.
Let me konw if you make some progresses.

gp1234567, 2022-06-09:
1. The new config result of 30 epoch as follows，the AP result is still lower than that of centerpoint, especially pedestrian and truck.
[image]
2. config as follows
model = dict(
    type='SST_CP_MULTIPILLAR',
    vfe=dict(
        type='PillarFeatureNet',
        voxel_size=voxel_size,
        pc_range=det_range,
        use_norm=True,
        num_filters=[128],
        with_distance=False,
    ),
    mid_encoder=dict(
        type='SSTInputLayerV2',
        window_shape=window_shape,
        sparse_shape=(248, 560, 1),
        shuffle_voxels=True,
        debug=True,
        drop_info=drop_info,
        pos_temperature=1000,
        normalize_pos=False,
    ),
    backbone=dict(
        type='SSTv2',
        d_model=[128, ] * 4,
        nhead=[8, ] * 4,
        num_blocks=4,
        dim_feedforward=[256, ] * 4,
        output_shape=[248, 560],
        num_attached_conv=4,
        conv_kwargs=[
            dict(kernel_size=3, dilation=1, padding=1, stride=1),
            dict(kernel_size=3, dilation=1, padding=1, stride=1),
            dict(kernel_size=3, dilation=1, padding=1, stride=1),
            dict(kernel_size=3, dilation=2, padding=2, stride=1),
        ],
        conv_in_channel=128,
        conv_out_channel=128,
        debug=True,
        conv_shortcut=True,
    ),
    neck=dict(
        type='SECONDFPN',
        in_channels=[128, ],
        upsample_strides=[1, ],
        out_channels=[384, ]
    ),
    head=dict(
        type='SSTHead',
        cfg=dict(
            task_names=class_names,
            grid_size=np.array(grid_size),
            box_loss_list=box_loss_list,
            in_channels=384,  # 128 + 128 + 128
            shared_conv_channel=64,

            head_cfg=dict(
                head_in_channels=64,
                head_conv=64,
                final_kernel=3,
                bn=True,

                heads=[
                    dict(
                        hm=[2, 3],
                        reg=[2, 3],
                        height=[1, 3],
                        dim=[3, 3],
                        rot=[2, 3],
                        vel=[2, 3],
                        ious=[1, 3],
                        state=[1, 3]
                    ),
                    dict(
                        hm=[2, 3],
                        reg=[2, 3],
                        height=[1, 3],
                        dim=[3, 3],
                        rot=[2, 3],
                        vel=[2, 3],
                        ious=[1, 3],
                        state=[1, 3]
                    )
                ],
            )
        ),
        loss=dict(
            code_weights=[
                [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.25, 0.25, 0.5, 0.5, 0.5],
                # [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.5, 0.5],
                [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.5, 0.5, 0.5],
            ],
            vel=True,
            motion_state=True,
            hm_weight=[2.0, 2.0],
            # hm_weight=[1.0, 1.0],
            loc_weight=[1.0, 0.5],
            # loc_weight=[2.0, 2.0],
            box_weight=[0.25, 0.0],
            # box_weight=[0.0, 0.0]
            iou_loss=True,
            ious_config=ious_config
        )
    )
)
3. Can you give me some comments? thanks.

Abyssaledge, 2022-06-09:
Could you show me the performance and config of the CenterPoint model? And it would be nice to upload the config file instead of directly posting the config here.

gp1234567, 2022-06-09:
1. cp_config as follow
[cp.txt](https://github.com/TuSimple/SST/files/8866373/cp.txt)
2. the ap of centerpoint model is
[image]

gp1234567, 2022-06-10:
1. sst_cp config is
[sst_cp.txt](https://github.com/TuSimple/SST/files/8875227/sst_cp.txt)
2. the sst_cp ap is
[image]

closed 2022-06-14 by gp1234567