tusen-ai/SST #39: waymo validation set Eval problem

xieyunjiao-holo, 2022-06-23:
Hello, dear friend, I completed the training of waymo datasets, but an error was reported in the verification part, showing that the file could not be found. I did not find the compute_detection_metrics_main related files under the SST folder.
How to solve this please?
Next, how to skip the training and just do the evaluation on the validation set? 

[image]

[image]

Abyssaledge, 2022-06-23:
`compute_detection_metrics_main` is built from WOD source code. Please follow their [instructions](https://github.com/waymo-research/waymo-open-dataset)
And the docs in MMDetection3D might help: https://mmdetection3d.readthedocs.io/en/latest/datasets/waymo_det.html.

Abyssaledge, 2022-06-24:
I am not sure what the `pkl`  refers to. MMDetection3D will automatically save a `pkl` file and convert it to `bin`, you do not need to  convert it yourself.

closed 2022-06-27 by xieyunjiao-holo

JuiceLemonLemon, 2023-06-29:
> 

@xieyunjiao-holo 
Hello, it's good to see you.
How do you implement model evaluation code, could you please share it to me?

Abyssaledge, 2023-06-29:
@JuiceLemonLemon  You could follow this to build the evaluation tool: https://github.com/Abyssaledge/waymo-open-dataset-master/blob/master/docs/quick_start.md#local-compilation-without-docker-system-requirements

Besides, I provide a faster version here: https://github.com/Abyssaledge/waymo-open-dataset-master, which produces the same results with WOD official repo but much faster.

JuiceLemonLemon, 2023-06-29:
> @JuiceLemonLemon You could follow this to build the evaluation tool: [...]

Thank you for your quick reply.
Bazel installation is too hard for me, our server cannot download from network, and have no JAVA. So I tried to install Bazel, but failed.

JuiceLemonLemon, 2023-07-05:
> 

> @JuiceLemonLemon You could follow this to build the evaluation tool: [...]

I have finished to build the evaluation tool and got the waymo evaluation results, thank you very much.