tusen-ai/SST #42: About configuration question

seonhoon1002, 2022-07-11:
Hello, I have two questions about your codes.
1. In comments on config files in the D5 series, "D5 in the config name means the whole dataset is divided into 5 folds, We only use one fold for efficient experiments."
So this means that it uses 20% of the waymo dataset?
2. There are 5dim in configs, what is the mean of 5dim in lidar?

Summary
1. What is the mean of D5 in config files?
2. What is the mean of 5dim in config files in lidar?

Abyssaledge, 2022-07-11:
1. D5 means that sample 1 frame every 5 frames, which builds a sub training dataset containing 20% data.
2. 5dim means that use intensity and elongation as extra point features in addition to xyz.

seonhoon1002, 2022-07-11:
And I have an additional question about D5.

When I analyzed between "waymo-3d-3class" and "waymoD5-3d-3class" in "confgs/_base_/datasets", It seems like that only difference between two codes is setting of "use_dim" in "train/test/eval_pipeline">'LoadPointsFromFile"

Does "use_dim" makes difference for D5 and D1?

[image]

Abyssaledge, 2022-07-11:
`use_dim` has nothing to do with `load_interval`. The names of data config file are a little bit confusing, so you should focus on the `load_interval` in a config file instead of its name.

seonhoon1002, 2022-07-11:
So does the below part control interval?
[image]

Abyssaledge, 2022-07-11:
Yes

seonhoon1002, 2022-07-11:
Aha! I see, Thanks!!

closed 2022-07-11 by seonhoon1002