tusen-ai/SST #45: CUDA input/output Error

seonhoon1002, 2022-07-21:
Hello, I have problem with your "sst_refactor" code 
I trained it "sst_waymoD5_1x_3class_8heads_v2.py" and modified number of epoch "12 -->24"
But, I had problem of CUDA input/output Error. 
It occured randomly(sometimes it run well for several epochs), so I couldn't find reason of error.

**My configuration is bellow:**
TorchVision: 0.10.0+cu111
OpenCV: 4.6.0
MMCV: 1.4.0
MMCV Compiler: GCC 9.4
MMCV CUDA Compiler: 11.1
MMDetection: 2.14.0
MMSegmentation: 0.14.1
MMDetection3D: 0.15.0+7925828

**My error messages are bellow:**
Traceback (most recent call last):
  File "tools/train.py", line 230, in <module>
    main()
  File "tools/train.py", line 220, in main
    train_model(
  File "/home/lsh/code/pinpoint-feedback/mmdet3d/apis/train.py", line 27, in train_model
    train_detector(
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/mmdet/apis/train.py", line 170, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 47, in train
    for i, data_batch in enumerate(self.data_loader):
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
    data.reraise()
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/torch/_utils.py", line 425, in reraise
    raise self.exc_type(msg)
OSError: Caught OSError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/mmdet/datasets/dataset_wrappers.py", line 151, in __getitem__
    return self.dataset[idx % self._ori_len]
  File "/home/lsh/code/pinpoint-feedback/mmdet3d/datasets/custom_3d.py", line 358, in __getitem__
    data = self.prepare_train_data(idx)
  File "/home/lsh/code/pinpoint-feedback/mmdet3d/datasets/custom_3d.py", line 154, in prepare_train_data
    example = self.pipeline(input_dict)
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/mmdet/datasets/pipelines/compose.py", line 40, in __call__
    data = t(data)
  File "/home/lsh/code/pinpoint-feedback/mmdet3d/datasets/pipelines/loading.py", line 419, in __call__
    points = self._load_points(pts_filename)
  File "/home/lsh/code/pinpoint-feedback/mmdet3d/datasets/pipelines/loading.py", line 394, in _load_points
    pts_bytes = self.file_client.get(pts_filename)
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 992, in get
    return self.client.get(filepath)
  File "/home/lsh/anaconda3/envs/PPFBv1/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 518, in get
    value_buf = f.read()
OSError: [Errno 5] Input/output error

I attached my specific configuration
[sst_waymoD1_2x_3class_8heads_v2.txt](https://github.com/TuSimple/SST/files/9160138/sst_waymoD1_2x_3class_8heads_v2.txt)

Abyssaledge, 2022-07-22:
Could you check if this happen when you use other models like PointPillars?

seonhoon1002, 2022-07-22:
Yes, It is ok for "PointPillar, Second" in mmdet3d for "waymo dataset" and "sst>sst_waymoD1_2x_3class_8heads_3f.py" your model. It is weird problem because when I debug it single-gpu there was no error. 
Do you know anything about this problem?

Abyssaledge, 2022-07-22:
My suggestions:
1. Check if there is similar issues in repo of MMDetection3D and MMCV. Because it seems not related to the model of SST according to your error information.
2. Dose it always happen to the same data sample or happen randomly?

seonhoon1002, 2022-07-22:
1 --> ok, I will check this issue in repo of MMDection3D again.
2 --> It happens randomly but often occurred
I think I should more invest in this problem myself. If I find the error reason, I will tell you what they are. 
Thanks : )

seonhoon1002, 2022-07-23:
I found why was this error occured
Our gpu and harddisk caused this problem.
After I changed gpus and formated harddisk, there are no problem. 
Thanks to care of my problem.

closed 2022-07-23 by seonhoon1002