tusen-ai/SST #47: kitti dataset train problem

S-Eternal-S, 2022-07-28:
Hello, author, I am using the kitti dataset to train a network, but I have the following problem:
 > Traceback (most recent call last):
 > File "/home/eternal/Project/SST-main/tools/train.py", line 232, in <module>
 >   main()
 >  File "/home/eternal/Project/SST-main/tools/train.py", line 228, in main
 >    meta=meta)
 >  File "/home/eternal/Project/SST-main/mmdet3d/apis/train.py", line 34, in train_model
 >    meta=meta)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/mmdet/apis/train.py", line 244, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 30, in run_iter
    **kwargs)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/mmcv/parallel/data_parallel.py", line 75, in train_step
    return self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/mmdet/models/detectors/base.py", line 248, in train_step
    losses = self(**data)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/eternal/Project/SST-main/mmdet3d/models/detectors/base.py", line 58, in forward
    return self.forward_train(**kwargs)
  File "/home/eternal/Project/SST-main/mmdet3d/models/detectors/voxelnet.py", line 90, in forward_train
    x = self.extract_feat(points, img_metas)
  File "/home/eternal/Project/SST-main/mmdet3d/models/detectors/dynamic_voxelnet.py", line 43, in extract_feat
    x = self.middle_encoder(voxel_features, feature_coors, batch_size)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/eternal/miniconda3/envs/openMMLab/lib/python3.7/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/eternal/Project/SST-main/mmdet3d/models/middle_encoders/sst_input_layer_v2.py", line 77, in forward
    voxel_info = self.drop_voxel(voxel_info, 2) # voxel_info is updated in this function
  File "/home/eternal/Project/SST-main/mmdet3d/models/middle_encoders/sst_input_layer_v2.py", line 135, in drop_voxel
    keep_mask_s0, drop_lvl_s0 = self.drop_single_shift(batch_win_inds_s0)
  File "/home/eternal/Project/SST-main/mmdet3d/models/middle_encoders/sst_input_layer_v2.py", line 107, in drop_single_shift
    bincount = torch.bincount(batch_win_inds)
RuntimeError: CUDA error: no kernel image is available for execution on the device`

I did a simple experiment to confirm that the problem is not with cuda:
> Python 3.7.13 (default, Mar 29 2022, 02:18:16) 
> [GCC 7.5.0] :: Anaconda, Inc. on linux
> Type "help", "copyright", "credits" or "license" for more information.
> import torch
> torch.cuda.is_available()
> True
> torch.zeros(1).cuda()
> tensor([0.], device='cuda:0')

The generated config:
> voxel_size = (0.16, 0.16, 4)
> window_shape = (12, 12, 1)
> drop_info_training = dict({
    0: dict(max_tokens=30, drop_range=(0, 30)),
    1: dict(max_tokens=60, drop_range=(30, 60)),
    2: dict(max_tokens=100, drop_range=(60, 100000))
})
drop_info_test = dict({
    0: dict(max_tokens=30, drop_range=(0, 30)),
    1: dict(max_tokens=60, drop_range=(30, 60)),
    2: dict(max_tokens=100, drop_range=(60, 100)),
    3: dict(max_tokens=144, drop_range=(100, 100000))
})
drop_info = ({
    0: {
        'max_tokens': 30,
        'drop_range': (0, 30)
[... 477 lines cut ...]
    weight_decay=0.05,
    paramwise_cfg=dict(custom_keys=dict(norm=dict(decay_mult=0.0))))
optimizer_config = dict(grad_clip=dict(max_norm=10, norm_type=2))
lr_config = dict(
    policy='cyclic',
    target_ratio=(100, 0.001),
    cyclic_times=1,
    step_ratio_up=0.1)
momentum_config = None
runner = dict(type='EpochBasedRunner', max_epochs=12)
checkpoint_config = dict(interval=1)
log_config = dict(
    interval=50,
    hooks=[dict(type='TextLoggerHook'),
           dict(type='TensorboardLoggerHook')])
dist_params = dict(backend='nccl')
log_level = 'INFO'
work_dir = '/home/eternal/Project/SST-main/work_dirs/sst_waymoD5_1x_3class_centerhead_My/'
load_from = None
resume_from = None
workflow = [('train', 1)]
device = 'cuda'
gpu_ids = range(0, 1)

What is the reason for the above problem? How can I fix this

Abyssaledge, 2022-07-29:
My suggestions:
1. Make sure you could train other models like PointPillars.
2. Using single GPU non-distributed training and CUDA_LAUNCH_BLOCKING=1 to locate the problem.

S-Eternal-S, 2022-08-01:
Hello, author. According to your suggestion, I solved the problem of code feedback and the network is able to train. But there is a very strange phenomenon during the training process, the network is constantly increasing the learning rate, and the loss has no downward trend. The dataset I use is the public kitti dataset, and I have also changed the network configuration file for some parameter requirements in the kitti dataset, but the training effect of the network is still not improved. Excuse me, is my parameter change wrong? Or some other problem
[image]
?

Abyssaledge, 2022-08-14:
Sorry for late reply. According to the log, the learning rate seems too large, which may lead to fail to converge.

closed 2022-09-21 by Abyssaledge