tusen-ai/SST #48: Configuration of Waymo Open Dataset

cjfghddl0206, 2022-07-29:
I downloaded the Waymo Open Dataset and converted it into KITTI format through the corresponding github(https://github.com/caizhongang/waymo_kitti_converter). But to make it similar to the KITTI format, I need the ImageSets folder(train.txt, test.txt, trainval.txt, val.txt) and I also need a .pkl file, but I don't know where to do this.

Abyssaledge, 2022-07-29:
You should follow MMDetection3D to convert the raw waymo data into KITTI format.
See https://mmdetection3d.readthedocs.io/en/latest/data_preparation.html

cjfghddl0206, 2022-07-29:
The hard disk capacity I have is 4TB, so I think it will exceed the capacity. Is it possible to convert the code using the waymo_kitti_converter github to the KITTI format and convert it to the remaining data file with tools/create_data.py?

Abyssaledge, 2022-07-30:
I have not tried the waymo_kitti_converter tool. 
Some tips for saving space:
1. You could remove the raw xx.tar files after extracting *.tfrecord files.
2. Modify the code to skip the processing of image data if you don't need.
3. Save the xx.bin file with half precision, but remember to convert them to float32 after you read them into memory.

closed 2022-08-14 by Abyssaledge