tusen-ai/SST #49: Could I get more detail result about FSD in Waymo

seonhoon1002, 2022-08-02:
I got many inspires from your paper.
And I want know results evaluated by range in Waymo, such as [0,30], [30,50], [50, inf] in LV2.
Could I get more detail results evaluated by range in LV2 AP/APH in FSD?

Abyssaledge, 2022-08-02:
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1: [mAP 0.794841] [mAPH 0.79032]
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2: [mAP 0.702686] [mAPH 0.698626]
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1: [mAP 0.836176] [mAPH 0.782221]
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2: [mAP 0.743251] [mAPH 0.693881]
OBJECT_TYPE_TYPE_SIGN_LEVEL_1: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_SIGN_LEVEL_2: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1: [mAP 0.753034] [mAPH 0.740749]
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2: [mAP 0.733174] [mAPH 0.721126]
RANGE_TYPE_VEHICLE_[0, 30)_LEVEL_1: [mAP 0.93294] [mAPH 0.929189]
RANGE_TYPE_VEHICLE_[0, 30)_LEVEL_2: [mAP 0.925811] [mAPH 0.92206]
RANGE_TYPE_VEHICLE_[30, 50)_LEVEL_1: [mAP 0.784553] [mAPH 0.779691]
RANGE_TYPE_VEHICLE_[30, 50)_LEVEL_2: [mAP 0.70535] [mAPH 0.700984]
RANGE_TYPE_VEHICLE_[50, +inf)_LEVEL_1: [mAP 0.584136] [mAPH 0.577155]
RANGE_TYPE_VEHICLE_[50, +inf)_LEVEL_2: [mAP 0.452022] [mAPH 0.446415]
RANGE_TYPE_PEDESTRIAN_[0, 30)_LEVEL_1: [mAP 0.870343] [mAPH 0.82887]
RANGE_TYPE_PEDESTRIAN_[0, 30)_LEVEL_2: [mAP 0.82494] [mAPH 0.784705]
RANGE_TYPE_PEDESTRIAN_[30, 50)_LEVEL_1: [mAP 0.826189] [mAPH 0.765765]
RANGE_TYPE_PEDESTRIAN_[30, 50)_LEVEL_2: [mAP 0.741001] [mAPH 0.685885]
RANGE_TYPE_PEDESTRIAN_[50, +inf)_LEVEL_1: [mAP 0.744849] [mAPH 0.655662]
RANGE_TYPE_PEDESTRIAN_[50, +inf)_LEVEL_2: [mAP 0.576472] [mAPH 0.505481]
RANGE_TYPE_SIGN_[0, 30)_LEVEL_1: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[0, 30)_LEVEL_2: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[30, 50)_LEVEL_1: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[30, 50)_LEVEL_2: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[50, +inf)_LEVEL_1: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[50, +inf)_LEVEL_2: [mAP 0] [mAPH 0]
RANGE_TYPE_CYCLIST_[0, 30)_LEVEL_1: [mAP 0.860924] [mAPH 0.849141]
RANGE_TYPE_CYCLIST_[0, 30)_LEVEL_2: [mAP 0.856909] [mAPH 0.845171]
RANGE_TYPE_CYCLIST_[30, 50)_LEVEL_1: [mAP 0.707714] [mAPH 0.695178]
RANGE_TYPE_CYCLIST_[30, 50)_LEVEL_2: [mAP 0.66755] [mAPH 0.655732]
RANGE_TYPE_CYCLIST_[50, +inf)_LEVEL_1: [mAP 0.574649] [mAPH 0.559819]
RANGE_TYPE_CYCLIST_[50, +inf)_LEVEL_2: [mAP 0.535058] [mAPH 0.52125]

seonhoon1002, 2022-08-02:
Thanks a lot

closed 2022-08-02 by seonhoon1002