tusen-ai/SST #51: how to realize the "Visualization of a SST detection sequence by AB3DMOT tracking"

sjtuljw520, 2022-08-10:
In reamde, there is a gif showing the "Visualization of a SST detection sequence by AB3DMOT tracking", anyone know how to do it?

Abyssaledge, 2022-08-10:
Could you be more specific and let us know which part you do not know how to deal with?

sjtuljw520, 2022-08-10:
Thanks for replying, can you sharing the code to do the visualization as below ?
[image]

Abyssaledge, 2022-08-10:
We use the visualization tools in [SimpleTrack](https://github.com/TuSimple/SimpleTrack#demo-and-api-example). Hope it helps.

sjtuljw520, 2022-08-10:
OK, thank you very much! @Abyssaledge

closed 2022-09-21 by Abyssaledge