tusen-ai/SST #52: The problem of dividing regions and offsets

Fengshihao1, 2022-08-13:
Hi, thanks for your great work!
I have some questions to ask:

1.         keep_mask_s0, drop_lvl_s0 = self.drop_single_shift(batch_win_inds_s0)
2. 
3.         drop_lvl_s0 = drop_lvl_s0[keep_mask_s0]
4.         voxel_keep_inds = voxel_keep_inds[keep_mask_s0]
5.         batch_win_inds_s0 = batch_win_inds_s0[keep_mask_s0]
6. 
7.         batch_win_inds_s1 = voxel_info['batch_win_inds_shift1']
8.         batch_win_inds_s1 = batch_win_inds_s1[keep_mask_s0]
9. 
10.         keep_mask_s1, drop_lvl_s1 = self.drop_single_shift(batch_win_inds_s1)
11. 
12.         drop_lvl_s0 = drop_lvl_s0[keep_mask_s1]
13.         voxel_keep_inds = voxel_keep_inds[keep_mask_s1]
14.         batch_win_inds_s0 = batch_win_inds_s0[keep_mask_s1]
15. 
16.         drop_lvl_s1 = drop_lvl_s1[keep_mask_s1]
17.         batch_win_inds_s1 = batch_win_inds_s1[keep_mask_s1]
  According to the above code，after the first area division, the number of voxels after padding and drop in each area is obtained. Why is the first keep_mask used in the second offset division area, and the keep_mask obtained in the second time should be applied to The top is divided for the first time?

Abyssaledge, 2022-08-14:
I do not fully understand your question. Would you mind making it more clear? It would be fine to use Chinese.
对于第一个问题: 为了去掉被扔掉的token, 在第二次划分之前会使用keep_mask_s0过滤一遍batch_win_inds_s1.
您后面的问题我没太理解．

Fengshihao1, 2022-08-14:
> I do not fully understand your question. Would you mind making it more clear? It would be fine to use Chinese. 对于第一个问题: 为了去掉被扔掉的token, 在第二次划分之前会使用keep_mask_s0过滤一遍batch_win_inds_s1. 您后面的问题我没太理解．

您好。非常感谢您对第一个问题的解答。针对第二个问题：第二次划分之前会进行过滤，但是第二次划分结束后，上述代码14行，会重新扔掉token,那这样我们第一次划分好的区域是不是就不够原有的30，60，100了？（代码12-17不是很理解）

Abyssaledge, 2022-08-14:
扔掉两次是为了降低实现的复杂度，这样之后两次attention的输入都是一样的．　30/60/100是指会用零值padding到30/60/100, 实际上有效的token数本来就是不到30/60/100的（并且这里还没有真正进行padding），所以再扔掉一点也无妨，最终在`get_flat2win_inds_v2`函数中还是会把它们padding到30/60/100的．

Fengshihao1, 2022-08-14:
> 扔掉两次是为了降低实现的复杂度，这样之后两次attention的输入都是一样的．　30/60/100是指会用零值padding到30/60/100, 实际上有效的token数本来就是不到30/60/100的（并且这里还没有真正进行padding），所以再扔掉一点也无妨，最终在`get_flat2win_inds_v2`函数中还是会把它们padding到30/60/100的．

明白了。非常感谢您的讲解！

closed 2022-09-21 by Abyssaledge