tusen-ai/SST #55: Occuring errors when training with 3 sweeps.

sjtuljw520, 2022-08-17:
I try to train the model with config file: configs/sst/sst_waymoD1_2x_3class_8heads_3f.py
but showing the errors as follows:

../mmdet3d/datasets/pipelines/loading.py", line 882, in __call__
    assert len(self.use_dim) >= 6

When I set the parameter "use_color" in config file "configs/_base_datasets/waymo-3d-3class-3sweep.py" as **false**,
the error still occurs.

Abyssaledge, 2022-08-17:
Do you modify the code? It should not happen because `use_color` in `LoadPointsFromFileResetLast` is False by default.

sjtuljw520, 2022-08-17:
I did not modify the code. It is wired, I have searched the "use_color" in the codebase, and did not see the use_color is modified to true anywhere.

sjtuljw520, 2022-08-17:
I try to print the value of "use_color", it shows that the value is "{'backend': 'disk'}". @Abyssaledge

sjtuljw520, 2022-08-17:
I have solved this problem by rewrite the init function of LoadPointsFromFileResetLast as follows:

def __init__(self,
                 coord_type,
                 load_dim=6,
                 use_dim=[0, 1, 2, 3],
                 tanh_dim=None, # to normalize intensity and elongation in WaymoOpenDataset
                 shift_height=False,
                 use_color=False,
                 file_client_args=dict(backend='disk'),
                 reset_value=0):
        super().__init__(
                 coord_type,
                 load_dim,
                 use_dim,
                 tanh_dim,
                 shift_height,
                 use_color,
                 file_client_args,
        )
        self.reset_value = reset_value

closed 2022-09-21 by Abyssaledge

Abyssaledge, 2022-10-28:
Thank you for report this bug. It is fixed now.