tusen-ai/SST #56: performance of centerpoint and PC-RCNN in your paper

seonhoon1002, 2022-08-21:
Hello, I have question about performance in FSD paper in table 1

Bellow is your table which guessed tested on Waymo  V1.2.0 validation
[image]

And this is the table in Centerpoint
[image]

And this is two version of PV-RCNN paper
[image]

I think CenterPoint was measured from Waymo V1.0

Could I ask how did you know what version of Waymo for applying the Centerpoint

Abyssaledge, 2022-08-21:
CenterPoint uses WOD >= 1.2. Note that Level 1 and Level 2 are different. Performance of Level 1 only takes GT containing more than 5 points into account.

seonhoon1002, 2022-08-21:
So if you're right, did centerpoint enter the wrong performacne about "PV-RCNN". 
Because PVRCNN performances in LV2 in centerpoint are **"vehicle: 65.4AP/64.8APH, pedestrian: 53.9AP/46.7APH"**
but performance in LV2 in original paper are **vehicle: 68.98AP/68.41APH, pedestrian: 66.04AP/57.61APH**

Abyssaledge, 2022-08-25:
Sorry for late reply. Both of them are right. The reported performance of PVRCNN in many papers are different because of different re-implementation and different version of dataset. I recommend to refer to the updated performance in PVRCNN paper.

seonhoon1002, 2022-08-25:
OK I see, Thank you

closed 2022-08-25 by seonhoon1002