tusen-ai/SST #57: 窗口划分的

Fengshihao1, 2022-08-27:
max_num_win_x = int(np.ceil((sparse_shape_x / win_shape_x)) + 1) # plus one here to meet the needs of shift.  40
max_num_win_y = int(np.ceil((sparse_shape_y / win_shape_y)) + 1) # plus one here to meet the needs of shift.  40
max_num_win_z = int(np.ceil((sparse_shape_z / win_shape_z)) + 1) # plus one here to meet the needs of shift.  2
max_num_win_per_sample = max_num_win_x * max_num_win_y * max_num_win_z  # 3200

batch_win_inds = coors[:, 0] * max_num_win_per_sample + \
                        win_coors_x * max_num_win_y * max_num_win_z + \
                        win_coors_y * max_num_win_z + \
                        win_coors_z 
作者，您好！
非常感谢您的工作！
我对窗口划分过程有一点疑惑。在您的SST工作中，根据点云范围以及Voxel大小可以求得sparse_shape大小为[468,468,1],在这个范围计算窗口数目的时候，x，y，z方向均加一是为了对点进行移位。这样求得每个体素对应的窗口号（batch_win_inds ）后，是不是可以理解为sparse_shape也变大了？这样还可以理解为single stride吗？

Abyssaledge, 2022-08-27:
可以理解为sparse shape变大了一个window的宽度，但是这只是在逻辑概念上变大，方便实现．实际特征以及特征分辨率是没有任何变化的，仍然是single stride．

Fengshihao1, 2022-08-27:
> 可以理解为sparse shape变大了一个window的宽度，但是这只是在逻辑概念上变大，方便实现．实际特征以及特征分辨率是没有任何变化的，仍然是single stride．

明白了。感谢您的解答！

closed 2022-09-21 by Abyssaledge

MWPony, 2023-12-12:
您好，我想问下，SST中的Region Shift仅仅只是对坐标加了shift，那是怎样做到的swin中的那种变尺度大小的窗口的效果呢？