tusen-ai/SST #67: Question about Dynamic voxelize

Fengshihao1, 2022-09-28:
作者，您好！
再次非常感谢您的工作！
我对动态体素化特征（Dynamic voxelize）的过程有一个问题。
我在测试其它的数据集的时候，代码 _Voxelization中
       ‘’ if max_points == -1 or max_voxels == -1:
            coors = points.new_zeros(size=(points.size(0), 3), dtype=torch.int)
            dynamic_voxelize(points, coors, voxel_size, coors_range, 3)
            return coors‘’
出现以下问题：
    dynamic_voxelize(points, coors, voxel_size, coors_range, 3)
RuntimeError: points must be contiguous。
一开始测试没有问题，测试了一段时间突然出现这种错误。
我也尝试过在points后加上''.contiguous()'',虽然这个错误解决了，但是后续的voxel encoder（DynamicVFE）的代码函数 def map_voxel_center_to_point(self, pts_coors, voxel_mean, voxel_coors)中：       
‘’ voxel_index = (
            pts_coors[:, 0] * canvas_z * canvas_y * canvas_x +
            pts_coors[:, 1] * canvas_y * canvas_x +
            pts_coors[:, 2] * canvas_x + pts_coors[:, 3])
        voxel_inds = canvas[voxel_index.long()]
        center_per_point = voxel_mean[voxel_inds, ...]‘’
出现索引超出范围的问题：
 in map_voxel_center_to_point：
    center_per_point = voxel_mean[voxel_inds, ...]
IndexError: index is out of bounds for dimension with size 0

请问出现这种情况的原因是哪一部分？

Abyssaledge, 2022-09-28:
可能是由于数值误差使得输入点云对应的voxel inds超出范围了, 可以尝试把data cfg中的point_cloud_range在3个方向上都稍微改小一点点。

Fengshihao1, 2022-09-28:
> 可能是由于数值误差使得输入点云对应的voxel inds超出范围了, 可以尝试把data cfg中的point_cloud_range在3个方向上都稍微改小一点点。

好的，我尝试一下。非常感谢您的回答。

Fengshihao1, 2022-09-28:
> 可能是由于数值误差使得输入点云对应的voxel inds超出范围了, 可以尝试把data cfg中的point_cloud_range在3个方向上都稍微改小一点点。

您好。我根据您的方法把点云范围调小以后还是会出现这个错误。训练的时候没有问题，测试的时候就会出现这种问题。刚开始测试没有问题，都是中途出现问题。而且数据集本身用其它实验测试过是没有问题的。

Abyssaledge, 2022-09-28:
可以先确定一下测试过程中是否对点云范围进行了过滤．

Fengshihao1, 2022-09-28:
> 可以先确定一下测试过程中是否对点云范围进行了过滤．

非常感谢您细心的讲解，已经找到问题所在，在对数据进行处理的时候，没有处理恰当。

closed 2022-10-13 by Abyssaledge

jb892, 2023-05-16:
> > 可以先确定一下测试过程中是否对点云范围进行了过滤．
> 
> 非常感谢您细心的讲解，已经找到问题所在，在对数据进行处理的时候，没有处理恰当。

hi，可以详细描述下数据处理的问题在哪儿吗？我也遇到相同的问题