tusen-ai/SST #69: How to fix point offset direction reverse in segmentation_head?

wutheringcoo, 2022-09-28:
I use points in bbox to predict offset form center of bbox, loss function is smothl1, most method is from segmentation_head, but i found offset direction is often reverse, How to fix this problem?
vis is below:
[image]

wutheringcoo, 2022-09-28:
I have check offset targets visualization, gt vis is below
ps: predicted offset in my model is trained in my dataset, i use your idea in my instance_head, picture above is trained 28epoch,   train dataset: 25000frames
[image]

Abyssaledge, 2022-09-28:
Do you change the model structure before segmentation_head? If you do, which part do you modify?

wutheringcoo, 2022-09-28:
backbone use my own model structure, instance_head predicts offset from your segmentation_head

closed 2022-10-13 by Abyssaledge