tusen-ai/SST #70: When I change the seg_voxel_size to train custom dataset, the voxel2point neck gives the following error,what does it mean?

XLMore, 2022-09-29:
[image]

Abyssaledge, 2022-09-29:
It means you may not correctly mask those points out of point_cloud_range.

XLMore, 2022-09-29:
> It means you may not correctly mask those points out of point_cloud_range.
I set the voxel_size and pc_range as follows:
seg_voxel_size = (0.16, 0.16, 0.22)
point_cloud_range = [-80, -32, -2.2, 80, 32, 4.4]

Abyssaledge, 2022-09-29:
How about the point_cloud_range in your data_cfg? The one in data_cfg is used  in PointsRangeFilter.

XLMore, 2022-09-29:
> PointsRangeFilter
point_cloud_range in data_cfg is the same as in waymo-fsd.py which is [-74.88, -74.88, -1.999, 74.88, 74.88, 3.999]，and the point_cloud_range in fsd_waymoD1_1x is also differenr from that in data_cfg, How can I change it accordingly?

XLMore, 2022-09-29:
> > PointsRangeFilter
> > point_cloud_range in data_cfg is the same as in waymo-fsd.py which is [-74.88, -74.88, -1.999, 74.88, 74.88, 3.999]，and the point_cloud_range in fsd_waymoD1_1x is also differenr from that in data_cfg, How can I change it accordingly?

point_cloud_range in data_cfg must be a sub_set of point_cloud_range outside,I'll give it a try

XLMore, 2022-09-29:
what is the small margin in z axis of point_cloud_range of data_cfg used for?

Abyssaledge, 2022-09-29:
You are supposed to change to point_cloud_range in data config to fit your dataset. The small margin makes the voxelization more safe.

XLMore, 2022-09-29:
I changed the point_cloud_range according to your advice and get the train loop run. But after some iterations,the loss curve looks very strange
[image]
[image]
Is it normal because it's still in the early epoches?

XLMore, 2022-09-29:
I think the sparse_shape in VoteSegmentor backnone is also affected by the point cloud range, but I have no clue, Do you have any advice?

Abyssaledge, 2022-09-29:
Note that we use `EnableFSDDetectionHookIter` hook, which means FSD model will first train segmentor for couple of iterations for warmup and then enable following detection part. So the early epochs may not effective.

closed 2022-10-13 by Abyssaledge