tusen-ai/SST #71: Installation

Dhagash4, 2022-09-29:
Hey! FIrst of all thanks for this amazing work and thanks a lot for providing open source code for the same. But I have query with installation. I have latest mm3ddetection running with pytorch 1.12.1 but if I try to install using the same versions of libraries library is not able to setup properly. Also the link to GETTING STARTED of mm3ddetection is expired. Besides that I have custom dataloader for my dataset, one question was if I want to use it for more than one class for example with nuscenes dataset is it possible to have a config for that?

Thanks a lot!

Abyssaledge, 2022-09-30:
1. Could you be more specific about the problem you occur during the installation?
2. All of our configs support multi-class detection. I do not fully understand your difficulty about the multi-class setting.

Dhagash4, 2022-09-30:
Thanks for the fast response, at first it was the issue with THC/THC.h they are torch headers which were removed in latest versions so had to downgrade Pytorch, and then all the installations issues were over and I was able to install SST. 

Now, since I am using my custom dataset, and I have modified the config, sst_base.py for multi-class  detection in my case classes are 7, and modified https://github.com/tusen-ai/SST/blob/main/configs/sst/sst_waymoD5_1x_3class_8heads.py (the following file according to my needs).

I am trying to run the code with following command
`python tools/train.py configs/sst/${MY_CONFIG}` 
and getting the following error.

`KeyError: 'FocalLoss is already registered in models`

My environment:
```
PyTorch=1.10.0+cu113
MMDetection=2.25.2
MMSegmentation=0.28.0
MMCV=1.4.0
MMdet3d=0.15.0 (installed using while running setup.py for SST repo)
```

Thanks a lot!

Abyssaledge, 2022-10-01:
I guess this error is caused by you using the different version of MMDet or MMSeg or MMCV. The focal loss has already been registered in one of them. I fix the problem now by force registration https://github.com/tusen-ai/SST/blob/main/mmdet3d/models/losses/focal_loss.py#L135

Dhagash4, 2022-10-04:
Thanks for the response and it fixed my issue, I am working on purely 3d data (custom dataset) which contains points (Nx7) and bboxs(Nx7[theta,cx,cy,cz,l,w,h]), the model training is working on SECOND network config so the dataloader and setup is working properly. As I am using my custom dataset I modified [this](https://github.com/tusen-ai/SST/blob/main/configs/sst/sst_waymoD1_2x_3class_8heads_3f.py) according to my dataset  and modified 
`POINT_CLOUD_RANGE=[0, -40, -2.5, 80, 40, 2.5]` and tried running the network, which resulted in following error

[error](https://github.com/tusen-ai/SST/files/9706486/log.txt)
[log_file](https://github.com/tusen-ai/SST/files/9706497/20221004_134403.log)

If I am changing `point_cloud_range` what else do I need to change?

Abyssaledge, 2022-10-07:
This error is caused by mismatch between. weight shape and data shape. Specifically, according to `(959x13 and 10x64)` in the error information, I guess the your input point cloud feature dimension is different with ours (Nx5). You could easily fix this issue by modifying the MLP channels in SIR or changing the input channels.

Dhagash4, 2022-10-07:
Thanks a lot for all the help, I will look into all the hyperparameters in detail but just one last question my input is quite sparse so there is this warning getting displayed

`No voxel belongs to drop_level:2 in shift 0`
 and it has something to do with drop_info_train, can you tell me briefly like what it is for and does it impact training at all?

Abyssaledge, 2022-10-07:
`No voxel belongs to drop_level:2 in shift 0` is just a hint, which does not affect the performance. It's okey to mute it or comment the print line.
It seems that you are running the older version SST. I recommend you use the configs in `sst_refactor`

Abyssaledge, 2022-11-04:
1. It's hard to say what leads to the NaN, while I suggest you disable fp16 training.
2. I find another issue in your config, you change the point cloud range but forget modifying sparse_shape in middle_encoder and output_shape in backbone.

closed 2022-11-17 by Dhagash4