tusen-ai/SST #73: installation and run problem

JessieW0806, 2022-10-02:
After when I run command run.sh, it gives the error as follows. 
importError: /mnt/cache/wangyingjie/SST/mmdet3d/ops/ball_query/ball_query_ext.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK2at10TensorBase8data_ptrIfEEPT_v

When install FSD, I refers to this: [https://github.com/tusen-ai/SST/issues/6](url)
My environment is as follows:

sys.platform: linux
Python: 3.8.13 | packaged by conda-forge | (default, Mar 25 2022, 06:04:18) [GCC 10.3.0]
CUDA available: True
GPU 0,1,2,3,4,5,6,7: NVIDIA A100-SXM4-80GB
CUDA_HOME: /usr/local/cuda
NVCC: Build cuda_11.2.r11.2/compiler.29618528_0
GCC: gcc (GCC) 5.4.0
PyTorch: 1.9.0+cu111
PyTorch compiling details: PyTorch built with:
  - GCC 7.3
  - C++ Version: 201402
  - Intel(R) Math Kernel Library Version 2020.0.0 Product Build 20191122 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v2.1.2 (Git Hash 98be7e8afa711dc9b66c8ff3504129cb82013cdb)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - NNPACK is enabled
  - CPU capability usage: AVX2
  - CUDA Runtime 11.1
  - NVCC architecture flags: -gencode;arch=compute_37,code=sm_37;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86
  - CuDNN 8.0.5
  - Magma 2.5.2
  - Build settings: BLAS_INFO=mkl, BUILD_TYPE=Release, CUDA_VERSION=11.1, CUDNN_VERSION=8.0.5, CXX_COMPILER=/opt/rh/devtoolset-7/root/usr/bin/c++, CXX_FLAGS= -Wno-deprecated -fvisibility-inlines-hidden -DUSE_PTHREADPOOL -fopenmp -DNDEBUG -DUSE_KINETO -DUSE_FBGEMM -DUSE_QNNPACK -DUSE_PYTORCH_QNNPACK -DUSE_XNNPACK -DSYMBOLICATE_MOBILE_DEBUG_HANDLE -O2 -fPIC -Wno-narrowing -Wall -Wextra -Werror=return-type -Wno-missing-field-initializers -Wno-type-limits -Wno-array-bounds -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-unused-result -Wno-unused-local-typedefs -Wno-strict-overflow -Wno-strict-aliasing -Wno-error=deprecated-declarations -Wno-stringop-overflow -Wno-psabi -Wno-error=pedantic -Wno-error=redundant-decls -Wno-error=old-style-cast -fdiagnostics-color=always -faligned-new -Wno-unused-but-set-variable -Wno-maybe-uninitialized -fno-math-errno -fno-trapping-math -Werror=format -Wno-stringop-overflow, LAPACK_INFO=mkl, PERF_WITH_AVX=1, PERF_WITH_AVX2=1, PERF_WITH_AVX512=1, TORCH_VERSION=1.9.0, USE_CUDA=ON, USE_CUDNN=ON, USE_EXCEPTION_PTR=1, USE_GFLAGS=OFF, USE_GLOG=OFF, USE_MKL=ON, USE_MKLDNN=ON, USE_MPI=OFF, USE_NCCL=ON, USE_NNPACK=ON, USE_OPENMP=ON, 

TorchVision: 0.10.0+cu111
OpenCV: 4.6.0
MMCV: 1.3.9
MMCV Compiler: GCC 7.3
MMCV CUDA Compiler: 11.1
MMDetection: 2.14.0+2028b0c
MMSegmentation: 0.14.1
MMDetection3D: 0.15.0

JessieW0806, 2022-10-02:
Looking forward to your reply!

Abyssaledge, 2022-10-02:
Thanks for using. Such `UndefinedSymbol` error is likely caused by incompatible library versions. But I am not sure what goes wrong.
Here is my log which contains the related information  hope it helps. https://github.com/tusen-ai/SST/files/9689623/sst_waymoD5_1x_3class_8heads_v2.log

JessieW0806, 2022-10-28:
@Abyssaledge , Thanks for replying. I have solved the problem. P.S. Could you please share the log of FSD? 
When I run FSD, the loss seems to be confused.
Thanks a lot!

2022-10-27 15:40:29,559 - mmdet - INFO - workflow: [('train', 1)], max: 12 epochs
2022-10-27 15:40:31.417923: I tensorflow/stream_executor/platform/default/dso_loader.cc:53] Successfully opened dynamic library libcudart.so.11.0
2022-10-27 15:42:09,909 - mmcv - INFO - Reducer buckets have been rebuilt in this iteration.
2022-10-27 15:45:26,896 - mmdet - INFO - Epoch [1][50/6587]	lr: 3.028e-05, eta: 5 days, 6:50:55, time: 5.781, data_time: 2.801, memory: 38188, loss_sem_seg: 0.0158, loss_vote: 0.9941, recall_Car: 1.0000, recall_Ped: 1.0000, recall_Cyc: 1.0000, num_clusters: 137.7200, num_fg_points: 395.5800, loss_cls.task0: 0.0087, loss_center.task0: 0.1709, loss_size.task0: 0.1327, loss_rot.task0: 0.0093, loss_cls.task1: 0.0152, loss_center.task1: 0.0000, loss_size.task1: 0.0000, loss_rot.task1: 0.0000, loss_cls.task2: 0.0168, loss_center.task2: 0.0000, loss_size.task2: 0.0000, loss_rot.task2: 0.0000, loss_rcnn_cls: 0.0199, num_pos_rois: 0.0000, num_neg_rois: 305.7800, loss_rcnn_bbox: 0.0000, loss_rcnn_corner: 0.0000, loss: 1.3834, grad_norm: 7.5680
2022-10-27 15:49:19,359 - mmdet - INFO - Epoch [1][100/6587]	lr: 3.115e-05, eta: 4 days, 18:21:40, time: 4.649, data_time: 2.951, memory: 38675, loss_sem_seg: 0.0141, loss_vote: 0.8855, recall_Car: 0.9995, recall_Ped: 1.0000, recall_Cyc: 1.0000, num_clusters: 140.4000, num_fg_points: 451.0800, loss_cls.task0: 0.0033, loss_center.task0: 0.6309, loss_size.task0: 0.2616, loss_rot.task0: 0.0712, loss_cls.task1: 0.0012, loss_center.task1: 0.0000, loss_size.task1: 0.0000, loss_rot.task1: 0.0000, loss_cls.task2: 0.0010, loss_center.task2: 0.0197, loss_size.task2: 0.0080, loss_rot.task2: 0.0054, loss_rcnn_cls: 0.0027, num_pos_rois: 0.0000, num_neg_rois: 40.4800, loss_rcnn_bbox: 0.0000, loss_rcnn_corner: 0.0000, loss: 1.9045, grad_norm: 20.9856
2022-10-27 15:53:03,610 - mmdet - INFO - Epoch [1][150/6587]	lr: 3.260e-05, eta: 4 days, 12:57:19, time: 4.485, data_time: 2.817, memory: 38962, loss_sem_seg: 0.0130, loss_vote: 0.8263, recall_Car: 0.9996, recall_Ped: 1.0000, recall_Cyc: 1.0000, num_clusters: 138.6600, num_fg_points: 497.5800, loss_cls.task0: 0.0098, loss_center.task0: 0.9201, loss_size.task0: 0.2554, loss_rot.task0: 0.1541, loss_cls.task1: 0.0016, loss_center.task1: 0.0651, loss_size.task1: 0.0284, loss_rot.task1: 0.0229, loss_cls.task2: 0.0006, loss_center.task2: 0.0187, loss_size.task2: 0.0108, loss_rot.task2: 0.0075, loss_rcnn_cls: 0.0008, num_pos_rois: 0.0000, num_neg_rois: 27.6800, loss_rcnn_bbox: 0.0000, loss_rcnn_corner: 0.0000, loss: 2.3353, grad_norm: 28.1011
2022-10-27 15:56:45,075 - mmdet - INFO - Epoch [1][200/6587]	lr: 3.464e-05, eta: 4 days, 9:54:58, time: 4.429, data_time: 2.722, memory: 38962, loss_sem_seg: 0.0120, loss_vote: 0.7882, recall_Car: 0.9999, recall_Ped: 0.9999, recall_Cyc: 1.0000, num_clusters: 141.7600, num_fg_points: 468.0000, loss_cls.task0: 0.0155, loss_center.task0: 1.0231, loss_size.task0: 0.2793, loss_rot.task0: 0.2181, loss_cls.task1: 0.0009, loss_center.task1: 0.0317, loss_size.task1: 0.0164, loss_rot.task1: 0.0103, loss_cls.task2: 0.0003, loss_center.task2: 0.0000, loss_size.task2: 0.0000, loss_rot.task2: 0.0000, loss_rcnn_cls: 0.0069, num_pos_rois: 0.1800, num_neg_rois: 93.4000, loss_rcnn_bbox: 0.8702, loss_rcnn_corner: 0.6817, loss: 3.9547, grad_norm: 64.2069
2022-10-27 16:00:52,097 - mmdet - INFO - Epoch [1][250/6587]	lr: 3.727e-05, eta: 4 days, 10:18:20, time: 4.940, data_time: 3.215, memory: 38962, loss_sem_seg: 0.0112, loss_vote: 0.7727, recall_Car: 0.9994, recall_Ped: 1.0000, recall_Cyc: 1.0000, num_clusters: 140.9200, num_fg_points: 468.5800, loss_cls.task0: 0.0127, loss_center.task0: 1.1399, loss_size.task0: 0.3146, loss_rot.task0: 0.2162, loss_cls.task1: 0.0011, loss_center.task1: 0.0378, loss_size.task1: 0.0196, loss_rot.task1: 0.0261, loss_cls.task2: 0.0004, loss_center.task2: 0.0102, loss_size.task2: 0.0072, loss_rot.task2: 0.0038, loss_rcnn_cls: 0.0044, num_pos_rois: 0.1000, num_neg_rois: 108.7200, loss_rcnn_bbox: 0.3129, loss_rcnn_corner: 0.2342, loss: 3.1249, grad_norm: 44.4770
2022-10-27 16:04:53,514 - mmdet - INFO - Epoch [1][300/6587]	lr: 4.047e-05, eta: 4 days, 10:07:59, time: 4.828, data_time: 3.132, memory: 39154, loss_sem_seg: 0.0103, loss_vote: 0.7433, recall_Car: 0.9997, recall_Ped: 1.0000, recall_Cyc: 1.0000, num_clusters: 147.6800, num_fg_points: 456.9000, loss_cls.task0: 0.0140, loss_center.task0: 1.0709, loss_size.task0: 0.2921, loss_rot.task0: 0.2233, loss_cls.task1: 0.0011, loss_center.task1: 0.0729, loss_size.task1: 0.0310, loss_rot.task1: 0.0319, loss_cls.task2: 0.0001, loss_center.task2: 0.0000, loss_size.task2: 0.0000, loss_rot.task2: 0.0000, loss_rcnn_cls: 0.0066, num_pos_rois: 0.1600, num_neg_rois: 99.5200, loss_rcnn_bbox: 0.3959, loss_rcnn_corner: 0.2303, loss: 3.1237, grad_norm: 40.9476
2022-10-27 16:09:00,389 - mmdet - INFO - Epoch [1][350/6587]	lr: 4.426e-05, eta: 4 days, 10:19:57, time: 4.938, data_time: 3.241, memory: 39154, loss_sem_seg: 0.0094, loss_vote: 0.7323, recall_Car: 0.9986, recall_Ped: 1.0000, recall_Cyc: 1.0000, num_clusters: 147.8600, num_fg_points: 431.1600, loss_cls.task0: 0.0118, loss_center.task0: 1.1022, loss_size.task0: 0.3047, loss_rot.task0: 0.1844, loss_cls.task1: 0.0008, loss_center.task1: 0.0195, loss_size.task1: 0.0065, loss_rot.task1: 0.0104, loss_cls.task2: 0.0003, loss_center.task [... cut ...] lusters: 3121.5200, num_fg_points: 118243.1200, loss_cls.task0: 0.0481, loss_center.task0: 0.7795, loss_size.task0: 0.1776, loss_rot.task0: 0.2519, loss_cls.task1: 0.0038, loss_center.task1: 0.3502, loss_size.task1: 0.1995, loss_rot.task1: 0.2540, loss_cls.task2: 0.0025, loss_center.task2: 0.3903, loss_size.task2: 0.1043, loss_rot.task2: 0.2529, loss_rcnn_cls: 0.0234, num_pos_rois: 37.5400, num_neg_rois: 3453.3800, loss_rcnn_bbox: 1.3983, loss_rcnn_corner: 0.5400, loss: 5.1556, grad_norm: 3.1290

Abyssaledge, 2022-10-28:
Here is the log [fsd_waymoD1_1x.log](https://github.com/tusen-ai/SST/files/9886382/fsd_waymoD1_1x.log).
I delete the log you posted, which too long to post here. Could you upload it as a file?

JessieW0806, 2022-10-28:
Sorry for that. Here is my log. The value of loss seems to be wrong. Could u please help me out?
[log.txt](https://github.com/tusen-ai/SST/files/9887820/log.txt)

Abyssaledge, 2022-10-28:
https://github.com/tusen-ai/SST/blob/main/configs/fsd/fsd_waymoD1_1x.py#L236
The batchsize you use is too large. You should scale the number (pos samples limit) along with your batchsize (~ 128 * batchsize)

JessieW0806, 2022-10-29:
Thanks for timely reply! I have changed this setting, however, the loss seems still be wrong.
INFO - Epoch [1][50/79041]	lr: 3.000e-05, eta: 8 days, 17:21:32, time: 0.795, data_time: 0.284, memory: 4435, loss_sem_seg: 0.0157, loss_vote: 0.8871, recall_Car: 1.0000, recall_Ped: 0.9800, recall_Cyc: 1.0000, num_clusters: 136.6600, num_fg_points: 427.8600, loss_cls.task0: 0.0096, loss_center.task0: 0.3746, loss_size.task0: 0.1888, loss_rot.task0: 0.0341, loss_cls.task1: 0.0154, loss_center.task1: 0.0000, loss_size.task1: 0.0000, 

The log txt
[log.txt](https://github.com/tusen-ai/SST/files/9892743/log.txt)
 is here. Looking forward to your reply.

Abyssaledge, 2022-10-30:
Could you please point out which part of config you have modified? It's not easy to carefully check the modification you have made according to your log.
And why you believe the loss is wrong?

JessieW0806, 2022-10-31:
1) I do not change any configs, just want to reproduce the result of FSD.
2) The loss starts at 1 and then goes up to 4. Besides, at the epoch 1, items such as loss_rcnn_bbox are 0.
Thanks a lot!

Abyssaledge, 2022-10-31:
1. It seems that you modify at least the number of GPUs according to your log, which changes the total batchsize and iterations.
Since FSD use syncBN and a iteration-based warmup, it will make a significant difference on performance. So I suggest you check what you modified exactly.
2. Loss increase is reasonable because we enable the detection part after 4000 iterations.
3.  Zero loss is also caused by you reducing the GPU number, so that 4000 iterations are not enough for a good segmentation warmup, and no foreground points are selected into the detection part.
4. I suggest users first go through the important parameters in configs before conducting experiments.

JessieW0806, 2022-11-24:
Thanks for your help!
I have another question. As you said "A hotfix is using our code to re-generate the waymo_dbinfo_train.pkl", I have processed Waymo data using the latest mmdet3d version.
Do I only need to re-generate only the waymo_dbinfo_train.pkl instead of re-processing the whole dataset?

Abyssaledge, 2022-11-24:
You only need to re-generate `waymo_dbinfo_train.pkl`. If you know well about the format, you only need to modify the coordinates in this pickle file instead of regeneration.
FYI, here are our pickles: https://share.weiyun.com/B3Ss4rid. You could compare them with your local data in case of unexpected bugs.

JessieW0806, 2022-11-24:
[log.txt](https://github.com/tusen-ai/SST/files/10084279/log.txt)
I have checked that my processed data are correct.
I also make sure the code is the same as your update.
However, the log file is still a lot different from your earlier reply (e.g., strange increase), could you please check it, please?

Abyssaledge, 2022-11-25:
You could try an experiment without dbsampler. I doubt there is something wrong with it.

JessieW0806, 2022-11-25:
[log2.txt](https://github.com/tusen-ai/SST/files/10088297/log2.txt)
I have tested the experiment without dbsampler yesterday, it seems to have the same error.

Abyssaledge, 2022-11-25:
Send me an email and I will offer you the trained checkpoint. You could use it for inference to see if the results match.

JessieW0806, 2022-11-25:
Thanks! yingjiewang@mail.ustc.edu.cn

Abyssaledge, 2022-11-25:
I have sent it to you.

JessieW0806, 2022-11-25:
[res.txt](https://github.com/tusen-ai/SST/files/10089542/res.txt)
The output results seem to be fine....
Why is the training of the FSD network from scratch not correct?

Abyssaledge, 2022-11-25:
It's hard to say what is going wrong. You could list the detailed procedure you followed, including gata generation, the code you use, and modifications you made if any. I will try to help.

JessieW0806, 2022-11-30:
Thanks for your reply! I re-install the environment, and the training process seems to be ok, but I still get some questions to ask for your opinion.
1) when I use the original config, the whole waymo dataset will be used. The training time is 4 days which is much slower than your log. I use the same setting as you provided with 8 A100 gpus and trained on a cluster 
srun -p ai4science --async --job-name=FSD_s --gres=gpu:8 --ntasks=8 --ntasks-per-node=8 --cpus-per-task=8 --kill-on-bad-exit=1 bash run2.sh
What do you guess is the reason for the difference in training time?

2) If I want to increase samples_per_gpu to 8 (for example), what else do I need to change to get the best performance?

3) I want to use part of the data set for experiments in the future (one-fifth), so it is OK？

Abyssaledge, 2022-12-09:
Sorry for late reply. How is it going now? 
1. I don't know, it's hard to say. Check the IO or timing to find the bottleneck.
2. Increase learning rate and increase the `num` in `IoUNegPiecewiseSampler` along with batch size.
3. If you realize this by setting the load_interval, it will be fine.