tusen-ai/SST #74: ImportError: cannot import name 'ball_query_ext'

pansanity666, 2022-10-09:
```  
File "/mnt/workspace/code/SST/mmdet3d/ops/ball_query/ball_query.py", line 4, in <module>
    from . import ball_query_ext
ImportError: cannot import name 'ball_query_ext'
```
Hi, I've got this error. Any idea? 
My environment:
``` mmcv-full==1.4.0
mmdet==2.25.2
mmsegmentation==0.28.0```

Best,

Abyssaledge, 2022-10-09:
I guess this error is caused by incorrect installation. You could refer to our [log](https://github.com/tusen-ai/SST/files/9689623/sst_waymoD5_1x_3class_8heads_v2.log) to check your installation.

pansanity666, 2022-10-09:
> I guess this error is caused by incorrect installation. You could refer to our [log](https://github.com/tusen-ai/SST/files/9689623/sst_waymoD5_1x_3class_8heads_v2.log) to check your installation.
problem solved by ``` python setup.py develop```

closed 2022-10-09 by pansanity666