tusen-ai/SST #75: question about fast run

pansanity666, 2022-10-09:
Hi, 
   The waymo dataset is too large. Is it possible to run the code without downloading the waymo dataset? (I only want to know the implementation details about the code, e.g., input and output shape of each function.)
Best,

Abyssaledge, 2022-10-09:
Yes, you could download a single waymo data file [here](https://console.cloud.google.com/storage/browser/waymo_open_dataset_v_1_3_1_individual_files;tab=objects?pli=1&prefix=&forceOnObjectsSortingFiltering=false).
After that we could provide a pretrianed model.

pansanity666, 2022-10-09:
> Yes, you could download a single waymo data file [here](https://console.cloud.google.com/storage/browser/waymo_open_dataset_v_1_3_1_individual_files;tab=objects?pli=1&prefix=&forceOnObjectsSortingFiltering=false). After that we could provide a pretrianed model.

Thank you for you reply! But the 'download' button on your provided website is gray and can not be clicked. Dont know specifically how to download this file. Is there download link from google drive etc?
Best,

Abyssaledge, 2022-10-09:
You could refer to the individual files in this [page](https://waymo.com/open/download/)

pansanity666, 2022-10-09:
> You could refer to the individual files in this [page](https://waymo.com/open/download/)

Thanks. The individual file can be downloaded after registration. 
Next, should I process the individual file following the same procedure as in mmdetection3d?

Abyssaledge, 2022-10-09:
I recommend to follow their documents but use the code in this repo to process the data, due to official MMDet3D update the coordinates system.

pansanity666, 2022-10-09:
> I recommend to follow their documents but use the code in this repo to process the data, due to official MMDet3D update the coordinates system.

Thanks. The mini-dataset has been ready by little modification on the data converter script. Well, where should I get the pre-trained checkpoint then? (If I have not miss something in README.md.)
Best,

Abyssaledge, 2022-10-09:
Just give me your email and I will send you a private link.

pansanity666, 2022-10-10:
***@***.***, thanks!

Sent Time:2022-10-10 07:51:30 (Monday)
To: tusen-ai/SST ***@***.***>
Cc: "Xiao Pan" ***@***.***>, Author ***@***.***>

Abyssaledge, 2022-10-10:
Could you show me the email at this page? The email address will be blocked if you reply via email.

pansanity666, 2022-10-10:
> Could you show me the email at this page? The email address will be blocked if you reply via email.
xiaopan@zju.edu.cn, Thank!

Abyssaledge, 2022-10-10:
I have sent you the link.

pansanity666, 2022-10-12:
> I have sent you the link

Appreciated!

closed 2022-10-12 by pansanity666