tusen-ai/SST #82: build TorchEx

Fengshihao1, 2022-11-26:
您好！clone https://github.com/Abyssaledge/TorchEx代码后，在编译的时候一直报错安装不成功。请问是有什么依赖包吗？我的环境是python=3.7，cuda10.2，pytorch1.9.1

报错如下：

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

Abyssaledge, 2022-11-26:
给个完整的编译方式和报错信息吧。

Fengshihao1, 2022-11-26:
编译方式： pip install -v .
Using pip 22.2.2 from /home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/pip (python 3.7)  

报错信息：
  ninja: build stopped: subcommand failed.
  Traceback (most recent call last):
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1672, in _run_ninja_build
      env=env)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/subprocess.py", line 512, in run
      output=stdout, stderr=stderr)
  subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

Traceback (most recent call last):
    File "<string>", line 36, in <module>
    File "<pip-setuptools-caller>", line 34, in <module>
    File "/mnt/data4/fengshihao/code/TorchEx-main/setup.py", line 47, in <module>
      'build_ext': BuildExtension
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/__init__.py", line 87, in setup
      return distutils.core.setup(**attrs)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/core.py", line 185, in setup
      return run_commands(dist)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
      dist.run_commands()
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
      self.run_command(cmd)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/dist.py", line 1208, in run_command
      super().run_command(command)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
      self.distribution.run_command(command)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/dist.py", line 1208, in run_command
      super().run_command(command)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/command/build.py", line 132, in run
      self.run_command(cmd_name)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
      self.distribution.run_command(command)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/dist.py", line 1208, in run_command
      super().run_command(command)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 84, in run
      _build_ext.run(self)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 346, in run
      self.build_extensions()
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 709, in build_extensions
      build_ext.build_extensions(self)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 468, in build_extensions
      self._build_extensions_serial()
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 494, in _build_extensions_serial
      self.build_extension(ext)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 246, in build_extension
      _build_ext.build_extension(self, ext)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 556, in build_extension
      depends=ext.depends,
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 539, in unix_wrap_ninja_compile
      with_cuda=with_cuda)
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1360, in _write_ninja_file_and_compile_objects
      error_prefix='Error compiling objects for extension')
    File "/home/fengshihao/anaconda3/envs/V2B_2/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1682, in _run_ninja_build
      raise RuntimeError(message) from e
  RuntimeError: Error compiling objects for extension
  error: subprocess-exited-with-error

Fengshihao1, 2022-11-26:
> 

我刚才用nvcc -V查看服务器cuda版本是7.5，但是nvidia-smi命令查出来的是10.2，是因为这个问题导致编译不成功吗？

Abyssaledge, 2022-11-26:
nvidia-smi显示的应该是驱动支持的最高cuda版本，你的真实cuda版本应该是7.5, 确实太低了。建议用10.2.

Fengshihao1, 2022-11-26:
> nvidia-smi显示的应该是驱动支持的最高cuda版本，你的真实cuda版本应该是7.5, 确实太低了。建议用10.2.

嗯，好的。非常感谢您的解答。

closed 2022-12-09 by Abyssaledge