tusen-ai/SST #89: The results of FSD on nuscenes or kitti

Pedrojj97, 2023-02-04:
Hi, have you ever tried the fsd model on nuscenes or kitti.

Abyssaledge, 2023-02-04:
Someone has tried FSD on KITTI: https://github.com/tusen-ai/SST/issues/77
The design of the nuScenes version is slightly different due to its evaluation metric being quite different from the WOD. We will release it in a couple of months.

Pedrojj97, 2023-02-04:
Thanks for your quickly replay, Would you mind telling me what the slightly difference is? In my opinion, fsd depends on the result of segmentation. Therefore, whether it is not very friendly to the datasets consist of many categories like nuscenes.

Abyssaledge, 2023-02-04:
It‘s a misunderstanding. FSD can well handle the Argoverse 2 dataset in our paper, which contains up to 30 classes.
Differences are:
1. nuScenes does not use IoU as a metric, so the label assignment in the GroupCorrection head should not use IoU.
2. In WOD, a point cloud could belong to more than 1 category (Binary sigmoid CE loss as segmentation loss). In nus or Argo, we restrict a point only belonging to a single class, so we use softmax CE loss as segmentation loss and the `group_sample` function to sample fg.

Pedrojj97, 2023-02-04:
I got it, thanks again~

closed 2023-02-04 by Pedrojj97