tusen-ai/SST #92: Configs of Argoverse and nuscenes

rentoumacsj, 2023-02-24:
Hi，Would you mind releasing the configs of Argoverse and nuscenes.

Abyssaledge, 2023-02-24:
Sorry, we are too busy now. We are planning to release these in the middle of March.

rentoumacsj, 2023-02-25:
Thanks for your replay,  Can you provide some suggestions on how to set the segmentation threshold of each category,  and how to set the group name and group length when use group_sample function on the datasets with more categories such as argoverse and nuscenes.

foolhard, 2023-03-17:
Hi, any update on Argoverse configuration?

Abyssaledge, 2023-03-21:
The Argoverse model is released now. See the instructions: https://github.com/tusen-ai/SST/blob/main/instructions.md
@rentoumacsj @foolhard

rentoumacsj, 2023-03-21:
Thanks for reminding me，It seems that the two-stage model is not used on Argoverse dataset.

foolhard, 2023-03-22:
@Abyssaledge Thanks for remiding me. 
I tried the config file and script and found an issue for testing:
`ModuleNotFoundError: No module named 'numpy.typing'`
I installed numpy==1.21.5 to solve the issue. 
However, mmdet3d==0.15.0 requires numpy < 1.20.0 and then training script fails. 

What's your environment for mmdet3d and numpy?

Abyssaledge, 2023-03-22:
In fact,  numpy=1.21.0 is ok in our repo. You can install NumPy 1.21.0 after installing mmdet3d.
Some warnings related to data type might appear after installing NumPy 1.21.0, you could mute them with some simple modifications, according to the warning information.
@foolhard

foolhard, 2023-03-22:
I met an issue of av2_api. 
It seems that they generated the wrong file path when loading map file. 

Do you meet this issue？How do you install av2_api?  

```
[Parallel(n_jobs=-1)]: Using backend ThreadingBackend with 32 concurrent workers.
Traceback (most recent call last):
  File "tools/test.py", line 215, in <module>
    main()
  File "tools/test.py", line 211, in main
    print(dataset.evaluate(outputs, **eval_kwargs))
  File "/dfs/data/open_mmlab/SST/mmdet3d/datasets/argo2_dataset.py", line 487, in evaluate
    eval_dts, eval_gts, metrics = evaluate(
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/av2/evaluation/detection/eval.py", line 141, in evaluate
    log_id_to_avm, log_id_to_timestamped_poses = load_mapped_avm_and_egoposes(log_ids, cfg.dataset_dir)
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/av2/evaluation/detection/utils.py", line 455, in load_mapped_avm_and_egoposes
    avms: Optional[List[ArgoverseStaticMap]] = Parallel(n_jobs=-1, backend="threading", verbose=1)(
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/joblib/parallel.py", line 1056, in __call__
    self.retrieve()
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/joblib/parallel.py", line 935, in retrieve
    self._output.extend(job.get(timeout=self.timeout))
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/multiprocessing/pool.py", line 771, in get
    raise self._value
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/joblib/_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/joblib/parallel.py", line 262, in __call__
    return [func(*args, **kwargs)
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/joblib/parallel.py", line 262, in <listcomp>
    return [func(*args, **kwargs)
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/av2/map/map_api.py", line 375, in from_map_dir
    static_map = cls.from_json(vector_data_json_path)
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/av2/map/map_api.py", line 325, in from_json
    vector_data = io_utils.read_json_file(static_map_path)
  File "/dfs/data/miniconda/envs/sst/lib/python3.8/site-packages/av2/utils/io.py", line 218, in read_json_file
    with open(fpath, "rb") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'data/argo2/argo2_format/sensor/val/02678d04-cc9f-3148-9f95-1ba66347dff9/map/data/argo2/argo2_format/sensor/val/02678d04-cc9f-3148-9f95-1ba66347dff9/map/log_map_archive_02678d04-cc9f-3148-9f95-1ba66347dff9____PIT_city_71109.json'
```

foolhard, 2023-03-22:
@Abyssaledge , 

It did work. 
I just check it in `av2_api` and they already fixed in 'main' branch. 

Thanks for your quick response and kindly help. 

> It seems a bug of av2-api, try the following instruction: In L681 of `av2/map/map_api.py`, change `log_map_dirpath / vector_data_frame` to `vector_data_frame` L681 might be not accurate if you install a different API version, but it is easy to find the real position by searching.