tusen-ai/SST #94: Question regarding reproducing SST results on waymo

richardkxu, 2023-03-07:
Hi, thank you for the great repo! I am trying to reproduce the SST results on waymoD5 dataset (`load_interval=5`) but got much lower metrics. Here are the results I got: 

`sst/sst_waymoD5_1x_3class_12heads.py` results:
```
Vehicle/L1 mAP: 0.4463, Vehicle/L1 mAPH: 0.4406, Vehicle/L2 mAP: 0.3903, Vehicle/L2 mAPH: 0.3853, Pedestrian/L1 mAP: 0.5345, Pedestrian/L1 mAPH: 0.4239, Pedestrian/L2 mAP: 0.4612, Pedestrian/L2 mAPH: 0.3651, Sign/L1 mAP: 0.0000, Sign/L1 mAPH: 0.0000, Sign/L2 mAP: 0.0000, Sign/L2 mAPH: 0.0000, Cyclist/L1 mAP: 0.3997, Cyclist/L1 mAPH: 0.3892, Cyclist/L2 mAP: 0.3848, Cyclist/L2 mAPH: 0.3747, 
Overall/L1 mAP: 0.4602, Overall/L1 mAPH: 0.4179, Overall/L2 mAP: 0.4121, Overall/L2 mAPH: 0.3750
```
which is lower than the Overall/L1 mAP: 0.6797 mentioned in [this issue](https://github.com/tusen-ai/SST/issues/11#issuecomment-1034439834).

`sst_refactor/sst_waymoD5_1x_3class_8heads_v2.py` results:
```
Vehicle/L1 mAP: 0.4355, Vehicle/L1 mAPH: 0.4299, Vehicle/L2 mAP: 0.3806, Vehicle/L2 mAPH: 0.3756, Pedestrian/L1 mAP: 0.5287, Pedestrian/L1 mAPH: 0.4172, Pedestrian/L2 mAP: 0.4561, Pedestrian/L2 mAPH: 0.3593, Sign/L1 mAP: 0.0000, Sign/L1 mAPH: 0.0000, Sign/L2 mAP: 0.0000, Sign/L2 mAPH: 0.0000, Cyclist/L1 mAP: 0.3911, Cyclist/L1 mAPH: 0.3804, Cyclist/L2 mAP: 0.3766, Cyclist/L2 mAPH: 0.3663, 
Overall/L1 mAP: 0.4518, Overall/L1 mAPH: 0.4092, Overall/L2 mAP: 0.4044, Overall/L2 mAPH: 0.3671
```
which is lower than the Overall/L1 mAP: 0.6734 mentioned in [this issue](https://github.com/tusen-ai/SST/issues/73#issuecomment-1264597764).

I am running with 8 GPUs and the exact same config files as provided in the repo. I am using the `mmdet==0.15.0` installed directly from your repo and the old version of waymo w/ old coordinate system generated from your `create_data.py`. I also run evaluation on some pretrained checkpoints and was able to get the same evaluation results on validation set. I think that partly verify my data and environment are correct.

Am I missing anything? Thank you so much!

Abyssaledge, 2023-03-11:
Sorry for the late reply.
According to 
`I also run evaluation on some pretrained checkpoints and was able to get the same evaluation results on validation set.`,
do you mean that you could obtain normal results using our checkpoint?

richardkxu, 2023-03-11:
Yes, I was able to get the FSD results using your checkpoint.

Abyssaledge, 2023-03-11:
How about training FSD from scratch? Does it work?

Abyssaledge, 2023-08-22:
Please reopen this issue if you need further discussion.

closed 2023-08-22 by Abyssaledge