tusen-ai/SST #102: Does CBGS Augmentation work in Argoverse2?

foolhard, 2023-04-05:
Hello, 

I try your CBGS augmentation below in standard CenterPoint but found it does not work. 
- Before 5 epoches, the validation mAP / CDS is better than w/o CBGS. But after 5 epochs, the mAP /CDS of CBGS are worse than w/o CBGS
- Training loss is still decreasing. 

Does CBGS help to improve detection score in AV2? Could you share your experiments results about CBGS in AV2?

Besides, I found that  RepeatedDataset is used in FSD, so FSD training doesn't re-sample the for class balance in AV2. Have you done the experiment for the class balance re-sampling in AV2? If yes, could you share the result?  

Here is the db_sampler I copy from FSD and used in standard centerpoint. 
```
db_sampler = dict(
    data_root=data_root,
    info_path=data_root + 'argo2_dbinfos_train.pkl',
    rate=1.0,
    prepare=dict(filter_by_difficulty=[-1], filter_by_min_points={k:5 for k in class_names}),
    classes=class_names,
    sample_groups=sample_groups,
    points_loader=dict(
        type='LoadPointsFromFile',
        coord_type='LIDAR',
        load_dim=4,
        use_dim=[0, 1, 2, 3],
        file_client_args=file_client_args))
```

Best regards

Abyssaledge, 2023-04-05:
I am not sure what "your CBGS augmentation" means, because I have not tried CBGS or other resamplings in AV2.
In my experience,  CBGS does not work that well in LiDAR-based detectors if we align the number of training iterations.
I believe CopyPaste and a longer training schedule might offer better performance than CBGS.

foolhard, 2023-04-06:
Thanks for quick feedback. 
Then I understood why you use RepeatedDatatset instead of CBGSDataset. 
Actually I mean the CopyPaste in [argo2-3d-26class.py](https://github.com/tusen-ai/SST/blob/main/configs/_base_/datasets/argo2-3d-26class.py). I use the same config in centerpoint but it does not work well. 
**How much does it improve your FSD in AV2?**

Abyssaledge, 2023-04-06:
Sorry, I have not trained FSD in AV2 without CopyPaste. What is the performance you got with CenterPoint?

closed 2023-08-22 by Abyssaledge