tusen-ai/SST #103: "DetectionCfg" problem  in Argo2Dataset

seonhoon1002, 2023-04-07:
Hello. I've been testing your model on av2.
But I have a problem DetectionCFG which is in "mmdet3d>datasets>argo2_dataset.py>Argo2Dataset line 481".

DetectionCfg in latest av2 library might has no "split" argument I think ...
Probably, I think this is the library version problem.
So Can I ask What version is your av2 libary?

Abyssaledge, 2023-04-07:
My av2 version is `0.1.0.`. Some users use `0.2.0`.

seonhoon1002, 2023-04-07:
I think that I can recover performances by deleting the below line in argo==0.2.0
https://github.com/tusen-ai/SST/blob/a42fcab3d5827492c95721b50c3871d47aa30335/mmdet3d/datasets/argo2_dataset.py#L481

Below  is the result that I tested
[image]

Abyssaledge, 2023-04-07:
Thanks for reminding me! Glad to see that!