tusen-ai/SST #105: will CTRL_FSD_TTA be open source?

20210726, 2023-04-23:
I'm interested in CTRL_FSD_TTA（https://waymo.com/open/challenges/entry/?challenge=DETECTION_3D&challengeId=DETECTION_3D&emailId=1cb154ab-1558&timestamp=1678678070312349），will it be open source?

Abyssaledge, 2023-04-23:
Yes we will release the code but it may take a couple of months since the paper is not public.

lcc815, 2023-05-06:
any updated plan to release the code?

Abyssaledge, 2023-05-07:
Thanks very much for your attention, please stay tuned patiently. If we are lucky with the submission, code release will be given the highest priority.

JuiceLemonLemon, 2023-06-08:
Hello, I'm aslo interested in CTRL. So I would like to know when will this code be released？

Abyssaledge, 2023-06-19:
@20210726  @lcc815 @JuiceLemonLemon Hi, CTRL is released now.

JuiceLemonLemon, 2023-06-20:
> @20210726 @lcc815 @JuiceLemonLemon Hi, CTRL is released now.

ok, thank you very much.

JuiceLemonLemon, 2023-06-26:
@Abyssaledge 
Hello,
Thank you for your sharing, and could you please tell me the password to get the resource in below link.

https://share.weiyun.com/1r6VsFfJ

closed 2023-06-28 by 20210726