tusen-ai/SST #112: wiil TCO module be open sourced?

slkxnw, 2023-06-20:
TCO improves the performance of CTRL, i'm interested in it. but it seems that tool/CTRL doesn't have C++ code, i wonder if it will be open sourced, or it's wroten by python?

Abyssaledge, 2023-06-20:
TCO is isolated from the detection part in this repo, and Yiming Mao is the developer of TCO. You could contact him using the email in our paper for more information.

slkxnw, 2023-06-20:
thank you very much! i will contact Mr. Mao later

closed 2023-06-20 by slkxnw

Joeless, 2023-07-13:
In TCO, the core operator is accelerated by CUDA. I will open source in late July. @slkxnw

slkxnw, 2023-07-24:
> In TCO, the core operator is accelerated by CUDA. I will open source in late July. @slkxnw

Thank you!

Furchtlos001, 2023-11-29:
@Joeless @Abyssaledge  @slkxnw   I would like to know if TCO is open sourced.

darnellMJ, 2024-05-08:
> In TCO, the core operator is accelerated by CUDA. I will open source in late July. @slkxnw

@Joeless @slkxnw Hello，if TCO is open sourced？I would like to communication about TCO. And I could not contact Mr. Mao using the email in paper.