tusen-ai/SST #117: docker build and run problem

yrt1993, 2023-06-26:
[problems.txt](https://github.com/tusen-ai/SST/files/11864031/problems.txt)

In the attached file are problems I met when I build docker and run the command in the CTRL_instructions.md. There is still several problems not solved:

1.  when I run` python ./tools/ctrl/generate_track_input.py ./tools/ctrl/data_configs/fsd_base_vehicle.yaml --process 8`

[image]

I want to know the mmcv-full version of yours.

2. when I run `python ./tools/ctrl/generate_train_gt_bin.py`

[image]

when I run `python ./tools/ctrl/extract_poses.py`

[image]

I want to know how to get the `./data` directory

Abyssaledge, 2023-06-26:
My environment is:
```
cumm-cu113                    0.3.4
mmcv-full                     1.3.9              
mmdet                         2.14.0
mmdet3d                       0.15.0            
mmsegmentation                0.14.1
spconv-cu113                  2.2.3
torch                         1.8.0+cu111
torchvision                   0.9.0+cu111
torch-scatter                 2.0.9
```
For the data, it is generated by the official MMDet3D script: `./tools/create_data.py`. For this part, you could follow the instructions of MMDet3Dv0.15. 

For `idx2timestamp.pkl`, please follow https://github.com/tusen-ai/SST/blob/main/docs/overall_instructions.md#fast-waymo-evaluation

Abyssaledge, 2023-06-26:
To be more clear, `waymo_format/training` directory contains raw waymo data (`xxx.tfrecord`).

yrt1993, 2023-06-27:
I got another problem when I run` python tools/create_data.py waymo --out-dir ./data/waymo`
[image]

it seems tensorflow version does not match with my cuda version. This is what I used, and I want to know the tensorflow-gpu and waymo-open-dataset-tf version of yours
[image]

Abyssaledge, 2023-06-27:
My tensorflow and waymo version is:
```
tensorflow                    2.4.0
waymo-open-dataset-tf-2-4-0   1.4.1
```
Besides, I usually follow the following shell script to setup my environment:
```
pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple
pip install tensorflow==2.4.0
pip3 install waymo-open-dataset-tf-2-4-0 --user
pip install torch-1.8.0+cu111-cp38-cp38-linux_x86_64.whl
pip install torchvision-0.9.0+cu111-cp38-cp38-linux_x86_64.whl
cd ./mmcv-1.3.9
TORCH_CUDA_ARCH_LIST="8.6" MMCV_WITH_OPS=1 pip install -e .
cd ..
pip install mmdet==2.14.0
pip install mmsegmentation==0.14.1
cd ./SST
pip install -e .
```
Hope it helps.

yrt1993, 2023-06-27:
and your cuda version?

Abyssaledge, 2023-06-27:
11.2

20210726, 2023-06-28:
what's your python version?

Abyssaledge, 2023-06-28:
3.8.x @20210726

CHXQ, 2023-07-07:
what's your numpy version?

atlantins, 2023-08-16:
can you show the folder ./data/waymo, is there will be two folder kitti format and waymo format?

atlantins, 2023-08-17:
can you show the pip list folder?  there is always some conflicts in my configuration
such as when i cd SST/ pip install -e.,  
tensorflow 2.4.0 requires typing-extensions~=3.7.4, but you have typing-extensions 4.7.1 which is incompatible.

Abyssaledge, 2023-08-17:
Tensorflow is always annoying. Just ignore the warning from TensorFlow, I guess it is not a big deal.

atlantins, 2023-08-24:
Can you tell me the space occupied after converting to kittiformat? I have 2.8T, but during the conversion process, libpng error: Write Error is displayed
[image]

Abyssaledge, 2023-08-25:
You could delete the raw `.tar` files to save the space. 2.8T should be enough.

atlantins, 2023-08-29:
[image]
When I generate my waymoopen, I often generate a directory larger than 2.8T when generating training. This is the directory generated by my training folder, and I have deleted some images. Our algorithm requires at least which folders. I don't know if you have encountered this problem,
and if i have my own waymo-open dataset ,how to generate "The detection results should be in official WOD submission format(.bin)" , through mmdetection to inference traning scenes to generate bin?how do you get it?Can you share your running command with me,
Thanks!

Abyssaledge, 2023-08-29:
Please open a new issue. Otherwise, I may forget to answer your question when I am free. Thanks! @atlantins

atlantins, 2023-08-29:
thanks a lot !  please see here https://github.com/tusen-ai/SST/issues/141