tusen-ai/SST #120: how to generate idx2timestamp.pkl and idx2contextname.pkl

yrt1993, 2023-06-28:
hi, I'm trying to run CRTL on [waymo_mini](https://github.com/open-mmlab/mmdetection3d/blob/main/docs/zh_cn/user_guides/dataset_prepare.md#waymo)，and I want to know how to generate these two pickles

Abyssaledge, 2023-06-28:
I recommend you first go through the instructions in `docs` for a quick start.  You could follow https://github.com/tusen-ai/SST/blob/main/docs/overall_instructions.md#fast-waymo-evaluation to use the two files.

20210726, 2023-08-15:
here is my code to generate train.txt val.txt test.txt and two pickles,hope it helps you:
`import os
from waymo_open_dataset.utils import range_image_utils
from waymo_open_dataset.utils import transform_utils
from waymo_open_dataset.utils import  frame_utils
from waymo_open_dataset import dataset_pb2 as open_dataset
from waymo_open_dataset.protos import metrics_pb2
from tqdm import tqdm
from ipdb import set_trace
import pickle as pkl

import tensorflow.compat.v1 as tf

idx2timestamp_dict={}
idx2contextname_dict = {}

data_folder = "./data/waymo/waymo_format/validation"
tf_records = os.listdir(data_folder)
tf_records = [x for x in tf_records if 'tfrecord' in x]
tf_records = sorted(tf_records)
bin_data = metrics_pb2.Objects()
print(bin_data)
file_path = "/media/hozon/nvme/dataset/waymo/train.txt"
file=open(file_path, "w")
bbb=0
for tf_record_name in tf_records:
    FILE_NAME = os.path.join(data_folder, tf_record_name)
#     print(FILE_NAME)
    dataset = tf.data.TFRecordDataset(FILE_NAME, compression_type='')
    context_name = tf_record_name.replace('segment-', '').replace('_with_camera_labels.tfrecord', '')
    count = 0
    print(bbb)
    for data in dataset:
        val = "1" + str(bbb).zfill(3) + str(count).zfill(3)
        count += 1
        file.write(val + "\n")
        frame = open_dataset.Frame()
        frame.ParseFromString(bytearray(data.numpy()))
        ts=frame.timestamp_micros
        idx2timestamp_dict[val] = ts
        idx2contextname_dict[val]=context_name
    bbb=bbb+1
file.close()

data_folder = "./data/dataset/waymo/waymo_format/testing"
tf_records = os.listdir(data_folder)
tf_records = [x for x in tf_records if 'tfrecord' in x]
tf_records = sorted(tf_records)
# print(tf_records)
bin_data = metrics_pb2.Objects()
print(bin_data)
file_path = "/media/hozon/nvme/dataset/waymo/test.txt"
file=open(file_path, "w")
bbb=0
for tf_record_name in tf_records:
    FILE_NAME = os.path.join(data_folder, tf_record_name)
#     print(FILE_NAME)
    dataset = tf.data.TFRecordDataset(FILE_NAME, compression_type='')
    context_name = tf_record_name.replace('segment-', '').replace('_with_camera_labels.tfrecord', '')
    count = 0
#     bbb = tf_record_name.split("_")
    print(bbb)
    for data in dataset:
        test = "2" + str(bbb).zfill(3) + str(count).zfill(3)
        count += 1
        file.write(test + "\n")
        frame = open_dataset.Frame()
        frame.ParseFromString(bytearray(data.numpy()))
        ts=frame.timestamp_micros
#         print(ts)
        idx2timestamp_dict[test] = ts
        idx2contextname_dict[test]=context_name

    bbb=bbb+1
file.close()

data_folder = "/media/hozon/nvme/dataset/waymo/waymo_format/training"
tf_records = os.listdir(data_folder)
tf_records = [x for x in tf_records if 'tfrecord' in x]
tf_records = sorted(tf_records)
bin_data = metrics_pb2.Objects()
print(bin_data)
file_path = "/media/hozon/nvme/dataset/waymo/train.txt"
file=open(file_path, "w")
bbb=0
for tf_record_name in tf_records:
    FILE_NAME = os.path.join(data_folder, tf_record_name)
    dataset = tf.data.TFRecordDataset(FILE_NAME, compression_type='')
    context_name = tf_record_name.replace('segment-', '').replace('_with_camera_labels.tfrecord', '')
    count = 0
    print(bbb)
    for data in dataset:
        train = "0" + str(bbb).zfill(3) + str(count).zfill(3)
        count += 1
        file.write(train + "\n")
        frame = open_dataset.Frame()
        frame.ParseFromString(bytearray(data.numpy()))
        ts=frame.timestamp_micros
        idx2timestamp_dict[train] = ts
        idx2contextname_dict[train]=context_name

    bbb=bbb+1
file.close()

with open('/media/hozon/nvme/dataset/waymo/idx2timestamp.pkl', 'wb') as fw:
    pkl.dump(idx2timestamp_dict, fw)
with open('/media/hozon/nvme/dataset/waymo/idx2contextname.pkl', 'wb') as fw:
    pkl.dump(idx2contextname_dict, fw)`

**This is the description of the Waymo dataset.**
https://github.com/open-mmlab/mmdetection3d/blob/0f9dfa97a35ef87e16b700742d3c358d0ad15452/docs/zh_cn/advanced_guides/datasets/waymo.md?plain=1#L96

Kampffussel03, 2024-06-11:
Thank you @20210726