tusen-ai/SST #123: Regarding how to train my own dataset using the CTRL method

luoxiaoliaolan, 2023-08-01:
First of all, I would like to express my sincere gratitude for your team's outstanding open-source work. I am highly interested in the CTRL method presented therein. In my previous work, I trained models using single-frame annotated point clouds, organized in a format similar to KITTI dataset. However, if I intend to train my own data using the CTRL method, I would need to provide annotated data for consecutive frames. Could you please advise me on how to organize the continuous frame data for training with the CTRL method? What information should be included (point clouds, poses, timestamps, categories)?

Moreover, I would appreciate any guidance on the key considerations when training my own dataset. If I obtain new training results, I will be more than happy to share them with you promptly. I am excited about the prospect of maintaining contact and engaging in further discussions. If convenient, you can add me on WeChat: lyb543918165. Looking forward to having more in-depth conversations with you.

luoxiaoliaolan, 2023-08-01:
@happynear @winstywang

Abyssaledge, 2023-08-01:
Thanks for your attention.
We have provided detailed instructions in docs. Since we also convert WOD to KITTI format, I believe it's not that hard to adapt the pipeline into your dataset. So I suggest you first go through our data processing tools in `./tools/ctrl/`. After that, I you have further specific questions, feel free to contact me (post here or by fanlue2019@ia.ac.cn). @luoxiaoliaolan

Abyssaledge, 2023-08-22:
Please reopen this issue if you need further discussion.

closed 2023-08-22 by Abyssaledge