tusen-ai/SST #128: mmcv error

xiaocongcsu, 2023-08-08:
Thanks for the author's great job~
I have a mmcv.ops import error when run the create_data.py command，did anybody meet the same problem? Thanks!
the detail error log is as follows:

(ctrl) ~/projects/model_zoo/SST$ python tools/create_data.py waymo --root-path ./data/waymo/ --out-dir ./data/waymo/
Traceback (most recent call last):
  File "tools/create_data.py", line 5, in <module>
    from data_converter import kitti_converter as kitti
  File "/home/projects/model_zoo/SST/tools/data_converter/kitti_converter.py", line 7, in <module>
    from mmdet3d.core.bbox import box_np_ops
  File "/home/projects/model_zoo/SST/mmdet3d/core/__init__.py", line 2, in <module>
    from .bbox import *  # noqa: F401, F403
  File "/home/projects/model_zoo/SST/mmdet3d/core/bbox/__init__.py", line 4, in <module>
    from .iou_calculators import (AxisAlignedBboxOverlaps3D, BboxOverlaps3D,
  File "/home/projects/model_zoo/SST/mmdet3d/core/bbox/iou_calculators/__init__.py", line 1, in <module>
    from .iou3d_calculator import (AxisAlignedBboxOverlaps3D, BboxOverlaps3D,
  File "/home/projects/model_zoo/SST/mmdet3d/core/bbox/iou_calculators/iou3d_calculator.py", line 5, in <module>
    from ..structures import get_box_type
  File "/home/projects/model_zoo/SST/mmdet3d/core/bbox/structures/__init__.py", line 1, in <module>
    from .base_box3d import BaseInstance3DBoxes
  File "/home/projects/model_zoo/SST/mmdet3d/core/bbox/structures/base_box3d.py", line 5, in <module>
    from mmdet3d.ops.iou3d import iou3d_cuda
  File "/home/projects/model_zoo/SST/mmdet3d/ops/__init__.py", line 21, in <module>
    from .sparse_block import (SparseBasicBlock, SparseBottleneck,
  File "/home/projects/model_zoo/SST/mmdet3d/ops/sparse_block.py", line 9, in <module>
    from mmcv.ops import SparseModule, SparseSequential
ImportError: cannot import name 'SparseModule' from 'mmcv.ops' (/home/anaconda3/envs/ctrl/lib/python3.8/site-packages/mmcv/ops/__init__.py)

And My environment is: 
mmcv-full                     1.3.9              
mmdet                         2.14.0
mmdet3d                       0.15.0            
mmsegmentation                0.14.1
spconv-cu113                  2.2.3
torch                         1.8.0+cu113
torchvision                   0.9.0+cu113

Abyssaledge, 2023-08-08:
If SpConv2.x is correctly installed, it will not import SparseModule from mmcv.
So you should figure out why `IS_SPCONV2_AVAILABLE` is not True:
https://github.com/tusen-ai/SST/blob/main/mmdet3d/ops/sparse_block.py#L6

xiaocongcsu, 2023-08-09:
> If SpConv2.x is correctly installed, it will not import SparseModule from mmcv. So you should figure out why `IS_SPCONV2_AVAILABLE` is not True: https://github.com/tusen-ai/SST/blob/main/mmdet3d/ops/sparse_block.py#L6

Thanks very much, it's been solved.

closed 2023-08-09 by xiaocongcsu