tusen-ai/SST #129: How to get the detection_results.bin?

xiaocongcsu, 2023-08-10:
Hi~Thanks for your great job
And I wondering how to get the waymo detection_results.bin?

> We recommand you to use CenterPoint(two-frame model for tracking) detection results for reproducing our results. Please follow https://github.com/tianweiy/CenterPoint or email its author for CenterPoint detection results.

Abyssaledge, 2023-08-10:
This is the instruction of the official ImmortalTracker repo. I do not have the CenterPoint detection results. I could provide you the result files of the FSD series detectors if you need.

xiaocongcsu, 2023-08-11:
> 

I‘ll appreciate it if you can send me the FSD series detectors results, my email is "xiaocongcsu@163.com".
Thanks again~

closed 2023-08-11 by xiaocongcsu

Abyssaledge, 2023-08-12:
OK, but I am currently inaccessible to our server, I will send it to you next week.
Feel free to remind me if I forget.
@xiaocongcsu