tusen-ai/SST #131: how to generate detection results for CTRL

yrt1993, 2023-08-15:
hi, I wonder how to generate detection results in the training step 2

https://github.com/tusen-ai/SST/blob/main/docs/CTRL_instructions.md#step-2-use-immortaltracker-to-generate-tracking-results-in-training-split-bin-file-format

Abyssaledge, 2023-08-16:
Using any detectors and inference on the Waymo dataset, then you can get the detection results. You could use the FSD in this repo if you need it.

Abyssaledge, 2023-08-22:
Please reopen this issue if you need further discussion.

closed 2023-08-22 by Abyssaledge

rockywind, 2023-08-23:
训练CTRL一定要有跟踪标签吗？
数据集可以是单帧的吗？

rockywind, 2023-09-11:
@Abyssaledge @yrt1993 
 您好，请问一下，FSDV2跑waymo数据集，如何存成bin格式的。
[image]